basler/pypylon #320: Basler MIPI - Problem using McmDevice: error 5461

ancabilloni, 2021-01-23:
Hello,

I'm using a bcon mipi daA2500-60mci on Jetson Nano.

**First of all**, this is the script that I use to read the camera. _(You can skip this part to read the error first on the next part)_

```
#! /usr/bin/env python3

from pypylon import pylon
import time
import cv2
import sys

basler_spec = {
        'res_h' : 1944, #pixels
        'res_w' : 2592 #pixels
        }

class BaslerCam(object):
    def __init__(self):
        self.camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
        self.camera.Open()
        self.action_active = False
        self.set_config()

    def set_config(self, auto_exposure=False, exposure_time=10000):
        if not auto_exposure:
            self.camera.ExposureAuto.SetValue("Off")
            self.camera.ExposureTime.SetValue(exposure_time)
        else:
            self.camera.ExposureAuto.SetValue("Continuous")

    def grabImage(self):
        self.action_active = True
        # converting to opencv bgr format
        converter = pylon.ImageFormatConverter()
        converter.OutputPixelFormat = pylon.PixelType_BGR8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
        self.camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

        # Grab as fast as possible
        while self.camera.IsGrabbing() and self.action_active:
            try:
                grabResult: pylon.GrabResult = self.camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
                if grabResult.GrabSucceeded():
                    # Access the image data
                    image = converter.Convert(grabResult)
                    self.img = image.GetArray()
                    #self.img = cv2.cvtColor( grabResult.GetArray(), cv2.COLOR_YUV2BGR_UYVY)
                    print(self.img.shape)
                grabResult.Release()
            except KeyboardInterrupt:
                break

        self.camera.StopGrabbing()
        self.camera.close()

    def close(self):
        self.action_active = False
        time.sleep(0.2)
        if self.camera.IsGrabbing():
            self.camera.StopGrabbing()
            self.camera.close()

def main(args):
    bc = BaslerCam()
    bc.grabImage()

if __name__ == '__main__':
    main(sys.argv)
```
**Second, the error part:**

Once the camera is warmed up and running, it will keep running how many times I reset the script above. However, after the Jetson is unplugged from power for at least 20 minutes, I will get this error every single time I start the script. And I'm not very sure what is the cause of this. 

```
python3 basler_mipi_cam.py
21.01.2021 22:12:15.092975 Didn't find a matching capture device for 'daA4200-30mci'
21.01.2021 22:12:15.094573 Didn't find a matching capture device for 'daA4200-30mci'
[+    0.000000][000000] WARNING: Problem using McmDevice: error 5461
[+    1.047440][000000] WARNING: Problem using McmDevice: read(): receive() failed: Read incomplete - requested bytes (8) != received bytes (65525)
Traceback (most recent call last):
  File "basler_mipi_cam.py", line 104, in <module>
    main(sys.argv)
  File "basler_mipi_cam.py", line 100, in main
    bc = BaslerCam()
  File "basler_mipi_cam.py", line 19, in __init__
    self.camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
  File "/home/user0/.local/lib/python3.6/site-packages/pypylon/pylon.py", line 1582, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: Cannot provide XML file. Failed to open GenTL device. : RuntimeException thrown (file 'PylonGtcTL.cpp', line 340)
```
My only remedy now is to read the camera from PylonViewer. The first frame in PylonViewer would return a black frame (I believe this indicate error state), but the second frame it will return a normal image color and the camera error state is fixed, and I can use the script above.

Did anyone have knowledge about this error? Can you advise a code setting solution if possible? Because I plan to use the Jetson as camera as embedded device without peripherals attached.

Thank you in advance!

stefanklug, 2021-02-11:
This issue was moved to different channel. I close this one here.

closed 2021-02-11 by stefanklug

ancabilloni, 2021-02-11:
hi @stefanklug where was this issue moved to?

ppurwar, 2021-11-05:
hi @ancabilloni. Were you able to fix this?
Please reopen this issue as I can not find it anywhere else.

barovehicles, 2022-02-10:
We have the same problem where was moved it?