basler/pypylon #322: Error when compiling setup.py source code

fmaldonadoos2017, 2021-01-26:
Hello, I have recently began working with the Basler acA1300-200uc (USB3) and when i run the setup.py code from the powershell, I get this error message below. Anyone know where i went wrong on the setup? is it a path issue? thank ahead. felix

(gg) PS D:\pypylon-master\pypylon-master> python setup.py
WARNING:root:git not found or invalid tag found.
WARNING:root:-> Building version from date!
Building version: 2021.1.26.dev53780
Remove: .\generated
Remove: .\pypylon
call D:\Anaconda3\envs\gg\Library\bin\swig.exe -python -outdir D:\pypylon-master\pypylon-master\pypylon -o D:\pypylon-master\pypylon-master\generated\genicam_wrap.cpp -c++ -Wextra -Wall -threads -modern -py3 -DHAVE_PYLON_GUI -D_WIN32 -D_WIN64 -IC:\Program Files\Basler\pylon 6\Development\include genicam.i
NodeCallback.i(166) : Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
IPort.i(23) : Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 6\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
PortImpl.i(52) : Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 6\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.

call D:\Anaconda3\envs\gg\Library\bin\swig.exe -python -outdir D:\pypylon-master\pypylon-master\pypylon -o D:\pypylon-master\pypylon-master\generated\pylon_wrap.cpp -c++ -Wextra -Wall -threads -modern -py3 -DHAVE_PYLON_GUI -D_WIN32 -D_WIN64 -IC:\Program Files\Basler\pylon 6\Development\include -IC:\Program Files\Basler\pylon 6\Development\include pylon.i
C:\Program Files\Basler\pylon 6\Development\include\pylon\gige\ActionTriggerConfiguration.h(72) : Warning 325: Nested struct not currently supported (ActionParameter ignored)
..\genicam\NodeCallback.i(166) : Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
..\genicam\IPort.i(23) : Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 6\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
..\genicam\PortImpl.i(52) : Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 6\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
C:\Program Files\Basler\pylon 6\Development\include\pylon\PylonImageBase.h(38) : Warning 403: Class 'Pylon::CPylonImageBase' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 6\Development\include\pylon\ReusableImage.h(50) : Warning 403: Method Pylon::IReusableImage::IsSupportedPixelType(Pylon::EPixelType) const might not be implemented.

Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\PylonCamEmu_v6_1_TL.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\PylonGigE_v6_1_TL.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\gxapi_v12.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy tree C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\pylonCXP => D:\pypylon-master\pypylon-master\pypylon\pylonCXP
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\PylonGUI_v6_1.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\PylonUtility_v6_1.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\PylonBase_v6_1.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\GCBase_MD_VC141_v3_1_Basler_pylon.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\GenApi_MD_VC141_v3_1_Basler_pylon.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\log4cpp_MD_VC141_v3_1_Basler_pylon.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\Log_MD_VC141_v3_1_Basler_pylon.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\NodeMapData_MD_VC141_v3_1_Basler_pylon.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\XmlParser_MD_VC141_v3_1_Basler_pylon.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\MathParser_MD_VC141_v3_1_Basler_pylon.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\PylonUsb_v6_1_TL.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\uxapi_v12.dll => D:\pypylon-master\pypylon-master\pypylon\
Copy C:\WINDOWS\System32\vcruntime140.dll => .\pypylon
Copy C:\WINDOWS\System32\msvcp140.dll => .\pypylon

usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied

thiesmoeller, 2021-01-26:
Either use the commands like in the Readme
https://github.com/basler/pypylon/blob/master/README.md#installation-from-source

Or call python setup.py install

lars-ek, 2021-01-27:
compiling with python 3.9 and pylon5 ok, but still an error on import:

~/pypylon $ python3
Python 3.9.1 (default, Jan  8 2021, 12:11:08)
[Clang 12.0.0 (clang-1200.0.32.28)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from pypylon import pylon
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/lib/python3.9/site-packages/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: dlopen(/opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so, 2): Symbol not found: __ZN5Pylon9CInfoBase16SetPropertyValueERKN29GenICam_3_1_Basler_pylon_v5_18gcstringES4_
  Referenced from: /opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so
  Expected in: flat namespace
 in /opt/

lars-ek, 2021-02-17:
I still get the error with the latest repo and have no idea what to do :(

 File "/opt/homebrew/lib/python3.9/site-packages/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: dlopen(/opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so, 2): Symbol not found: __ZN5Pylon9CInfoBase16SetPropertyValueERKN24GenICam_3_1_Basler_pylon8gcstringES4_
  Referenced from: /opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so
  Expected in: flat namespace
 in /opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so File "/opt/homebrew/lib/python3.9/site-packages/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: dlopen(/opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so, 2): Symbol not found: __ZN5Pylon9CInfoBase16SetPropertyValueERKN24GenICam_3_1_Basler_pylon8gcstringES4_
  Referenced from: /opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so
  Expected in: flat namespace
 in /opt/homebrew/lib/python3.9/site-packages/pypylon/_pylon.cpython-39-darwin.so

thiesmoeller, 2021-02-17:
We have macos binaries now

thiesmoeller, 2021-02-17:
Is this under WSL?

lars-ek, 2021-02-18:
macos 11.1 on m1-processor and python 3.9. No binary available:

~ $ pip3 install pypylon
ERROR: Could not find a version that satisfies the requirement pypylon
ERROR: No matching distribution found for pypylon
~ $

thiesmoeller, 2021-02-18:
As we have no 11.1 pylon yet, I assume you have to stay in the intel domain on rosetta2

Your homebrew is native arm?

lars-ek, 2021-02-18:
yes:

~ $ brew --config
HOMEBREW_VERSION: 3.0.1
ORIGIN: https://github.com/Homebrew/brew
HEAD: c951be8d3c7b339c7e759b7c40aec859e09a70a5
Last commit: 7 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 0b9a7d7295ad64b5aebb00e6413e2f85aaa71b20
Core tap last commit: 2 hours ago
Core tap branch: master
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 2.6.3 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 12.0 build 1200
Git: 2.30.1 => /opt/homebrew/bin/git
Curl: 7.64.1 => /usr/bin/curl
macOS: 11.2.1-arm64
CLT: 12.4.0.0.1.1610135815
Xcode: N/A
Rosetta 2: false

thiesmoeller, 2021-02-22:
have you tried to install an intel-homebrew on your apple-silicon?

https://noahpeeters.de/posts/apple-silicon/homebrew-setup/#install-an-intel-homebrew

barliya, 2022-01-07:
Hi,

I am facing the exact same problem with my M1 machine.
I have both silicon and intel versions of homebrew. 
But I don't see how this should help, I am using the pip3 to install not brew.
I am also running the terminal using Rosetta, and still not working.

Can you walk us through on how to install pypylon on m1 machines?

Thanks,
Avi

thiesmoeller, 2022-01-11:
Please post your command to install
And the output you get.

barliya, 2022-01-13:
Hi,
For now I have managed to make it work under python 3.8 installed in a virtual environment (using pipenv).
So it is working for me. 
However, I don't want it to be in the virtual env, so what I will try to do, is to downgrade my python from 3.9 to 3.8 and see if it also works for me in genral.
I will report back.
Thanks for the willingness to help.

thiesmoeller, 2023-06-14:
Release 2.2.0 with full support of macOS x86_64 and arm64 ( >= macOS 11 ) is out.

Available via pip install or releases page.

closed 2023-06-14 by thiesmoeller