basler/pypylon #326: Camera triggering projector

niallodowd, 2021-02-04:
Hello, I'm setting up a fringe projection system. I'm looking to trigger my projector with my camera (ace acA4600-7gc Color GigE). After setting up the circuit to use the opto-coupled output line, I'm not sure how to send the trigger using python script. Should this happen automatically, every time the camera grabs an image?

Please provide some direction, or a command which sends a pulse through the opto-coupled output line.

Thanks!
Niall

thiesmoeller, 2021-02-04:
It's called timer output. 
It gets started with start of exposure. 
https://docs.baslerweb.com/timer#ace-classicul-cameras

niallodowd, 2021-02-04:
Thanks so much for the quick response, Thies! Does this mean that the camera outputs the trigger to the opto-coupled line automatically? Or do I have to set up this functionality myself?

niallodowd, 2021-02-15:
I am trying this for example:

`camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
#%%
camera.Open()
camera.LineSelector.SetValue('Output Line 1')
camera.Close()`

but with no luck. Is there a way I can change these camera attributes?

niallodowd, 2021-02-15:
I'm trying to set the LineSelector by calling:

`camera.Open()

camera.LineSelector.SetValue('OutputLine1')

camera.Close()

`

however, I'm getting the error: **Feature 'LineSelector' : cannot convert value 'OutputLine1', the value is invalid**, though in pylon viewer, I am able to set this field to "Output Line 1".

thiesmoeller, 2021-02-15:
use the "Feature Documentation" window in pylon viewer to get the correct code for your setting.

only do a tiny translation to python 

e.g.: ExposureMode_Timed -> "Timed"
[image]

thiesmoeller, 2025-12-15:
This issue about triggering a projector from a camera has been addressed in the comments with guidance on GPIO output lines. We are closing this issue. For further questions on GPIO outputs, please refer to the Pylon SDK documentation or open a new issue.