basler/pypylon #327: Version error, can't run startup.py

fmaldonadoos2017, 2021-02-05:
I have python 3.6.1 installed, have pylon version 6.1.1 (image), and pypylon 1.6.0 installed with `pip install pypylon`. I have ran 
[image]. 

when calling 'setup.py' I get the message: 

```
WARNING:root:git not found or invalid tag found.
WARNING:root:-> Building version from date!
Building version: 2021.2.4.dev73181
Remove: .\generated
Remove: .\pypylon
```

I also get this one at the last line: 

`'The system cannot find the path specified: 'C:\\Program Files\\Basler\\pylon 6\\Development\\..\\runtime\\x64\\pylonCXP''
`
Its weird cause I can run grab.py from samples and see output, but some others dont work. For example when I try to run this in my code: 

`from samples.configurationeventprinter import ConfigurationEventPrinter
from samples.cameraeventprinter import CameraEventPrinter`

my question: are all my versions right? I feel like it might (or might not?) be a three-way compatibility issue? is there a specific python version + specific pypylon version + specific pylon version that needs to happen? 

Thanks in advance

thiesmoeller, 2021-02-07:
https://github.com/basler/pypylon/issues/203#issuecomment-585048892

thiesmoeller, 2021-02-23:
Beginning with release 1.7.1 we have binary releases on all platforms, that you can install via pip.

update your pip version before by calling `python -m pip install --upgrade pip`

closed 2021-02-23 by thiesmoeller