basler/pypylon #328: Basler pylon Gstreamer for python

mgabell, 2021-02-12:
Hi,

I have seen the library InstantCameraAppSrc library from Imaginghub that can be used to utilize Gstreamer and Basler camera. It does not however consider python. Its all C code. Anyone here who knows about a libs that support pypylon/python and Gstreamer?

br,
Mgabell

mgabell, 2021-02-23:
> @mgabell<https://github.com/mgabell>
> It works this way:
>
> +------------+        +--------------+     +-------------+
>
> |            |        |              |     |
>
> |   PYPYLON  +------->+   appsrc     +----->     gst
>
> |            |        |              |     |     pipeline..
>
> +------------+        +--------------+     +-------------+
>
> see this explanation:
> http://lifestyletransfer.com/how-to-use-gstreamer-appsrc-in-python/

Hi,

Thanks for the response. That means that you will go to the CPU buffer and back into (whatever GPU buffer of sorts)
which would decrease performance?

Br,
Magnus

Christina-ko, 2021-12-17:
> 

Hi, 

Can you provide me a guide on how to do the installation of GStreamer in ubuntu 18.04.
I followed the steps stated in https://github.com/joshdoe/gst-plugins-vision.
However, when I try this command:
gst-launch-1.0 pylonsrc ! autovideoconvert ! autovideosink

I got the error:
WARNING: erroneous pipeline: no element "pylonsrc"

Do you have any ideas what caused this error?

MagnusGabell, 2021-12-17:
> 

Hi, I dont know exactly. I didnt look at running gstreamer from console. But using python you can get it to work with below code. I will check more if I actually have a gst-launch that works. I used a basler USB3 camera. 
Ensure you have pylon installed. Also g-streamer, ofcourse. Then for the code below you also need pypylon. For windows I dont know how to install and get this to work, but in linux it should. not a direct answer to your question but I also struggle with the pipelines.
[simple_gstr.zip](https://github.com/basler/pypylon/files/7733227/simple_gstr.zip)

MagnusGabell, 2021-12-17:
> Hi,
>
> Can you provide me a guide on how to do the installation of GStreamer in ubuntu 18.04. I followed the steps stated in https://github.com/joshdoe/gst-plugins-vision. However, when I try this [...]

Try this below to ensure you have the pylon plugins installed properly check also the follow thread on the matter. I had the same problem as you... : https://github.com/joshdoe/gst-plugins-vision/issues/50

gst-inspect-1.0 | grep pylon

Christina-ko, 2021-12-17:
> > 
> 
> Hi, I dont know exactly. Ensure you have pylon library installed. Also g-streamer, of course. Then for the code attachedyou also need pypylon. For windows I dont know how to install and get this to work, but in linux it should. not a direct answer to your question but I also struggle with the pipelines. [simple_gstr.zip](https://github.com/basler/pypylon/files/7733227/simple_gstr.zip)

Hi, 

Thanks for the reply and sharing.

I have installed the pylon 6.2 in my Ubuntu 18.04  PC. 
How can I check whether my g-streamer is successfully installed in my PC?

MagnusGabell, 2021-12-17:
Check if you have the gst plugin. If gst-launch works responds and gst-inspect you probably has it. More likley the pylon-plugin is not.

//Magnus

Christina-ko, 2021-12-17:
> If gst-launch works responds and gst-inspect you probably have it. More likely the pylon-plugin is not.
> 
> //Magnus

Thank you 
Do you have any document on how to install g-streamer correctly in ubuntu 18.04 ?
Do you think this document is useful (https://gitlab.com/zingmars/gst-pylonsrc)?

thiesmoeller, 2021-12-17:
Check that GST_PLUGIN_PATH for local install or for a systemwide
install GST_PLUGIN_SYSTEM_PATH
is pointing to where you installed pylonsrc

Christina-ko, 2021-12-17:
> Check that GST_PLUGIN_PATH for local install or for a systemwide install GST_PLUGIN_SYSTEM_PATH is pointing to where you installed pylonsrc Christina-ko ***@***.***> schrieb am Fr., 17. Dez. 2021, 08:25:
> […](#)

Hi, 

I will check it and try it again. Thank you for your help

mgabell, 2021-12-17:
I found that it was easy to follow the instruction to get gstreamer to work but to get pylon to work was tricky. I think the solution was in the thread I referred to.