basler/pypylon #331: FPS independent of frames resolution and higher than expected

FraPochetti, 2021-02-17:
Hi, we are testing [this](https://www.baslerweb.com/en/products/cameras/area-scan-cameras/ace/aca2500-60um/) Basler camera and we are having issues understanding how images are grabbed and processed by `pypylon`.

Specifically, we try to estimate the camera FPS by fetching 1000 frames, measuring the requested `time`, and then computing `1000/time`. This approach results in ~195 FPS, against the 60 advertised. Those numbers look fishy, especially because they don't change when we adjust the camera resolution. We have tried with both `160x120` and `2590x2048` (native resolution).

Any ideas of what we might be doing wrong?

Here the snippet we are using:
```python
from pypylon import pylon
import time
import cv2

width, height, frames = 160, 120, 1000

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.Width.SetValue(width)
camera.Height.SetValue(height)
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

i=0
start_all = time.perf_counter()

while camera.IsGrabbing() and i<frames:
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        image = converter.Convert(grabResult)
        img = image.GetArray()
   
    assert grabResult.GrabSucceeded() == True
    grabResult.Release()
    i+=1

elapsed_all: float = time.perf_counter() - start_all    
camera.StopGrabbing()
camera.Close()

print(f"dim={width}x{height} fps={frames / elapsed_all :.2f} "
      f"time={elapsed_all:.2f} frames={frames}")
```

thiesmoeller, 2021-02-18:
The camera will calculate the FPS from all configured parameters.
The current active value is  represented  in this feature

https://docs.baslerweb.com/resulting-frame-rate

FraPochetti, 2021-02-18:
Thanks for the reply.

closed 2021-02-18 by FraPochetti