basler/pypylon #332: camera hardware triggering

niallodowd, 2021-02-19:
hello,

I've been struggling with hardware triggering my gigE acA4600. I can seldom take 60 images without missing an input trigger. I think that there may be some bug in my setup of the camera, or that I'm not using best practices, because the dropped frames occur at different indices, asking it impossible to code around. Here is my code:

`
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

frame_list=[]
camera.Open()
camera.TriggerMode.SetValue("On")
camera.TriggerActivation.SetValue("FallingEdge")
camera.TriggerSource.SetValue('Line1')
show=np.zeros([820,1150])
tot_frames=0
num_frames=0
pose_num=0
total=300
while True:
    k=cv2.waitKey(1)
    if k%256 == 27:
        print('esc pressed, exiting...')
        break
    cv2.imshow('frame',show)

    if k%256 == 32:
        print('grabbing started!')
        camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
        while camera.IsGrabbing():
            grabResult = camera.RetrieveResult(100000, pylon.TimeoutHandling_ThrowException)
            if grabResult.GrabSucceeded():        
                image = converter.Convert(grabResult)
                frame = image.GetArray()#[int(real_r[1]):int(real_r[1]+real_r[3]), int(real_r[0]):int(real_r[0]+real_r[2])]
                frame_list.append(frame)
                num_frames+=1
                tot_frames+=1
                print('pose: %s, just captured %s frames'%(pose_num,num_frames))
                cv2.imshow('frame',cv2.resize(frame,(1151,821))) 
                k=cv2.waitKey(2)
                if k%256 == 27:
                    k=cv2.waitKey(2)
                    print('esc pressed, exiting...')
                    break
                if num_frames==total:
                    print('ready')
                    pose_num+=1
                    num_frames=0
                    camera.StopGrabbing()
                    break
grabResult.Release()
# When everything done, release the capture
cv2.destroyAllWindows()
camera.Close()`

Pease advise, for this is causing substantial issues in my life....

PolymaTh-EE98, 2021-02-19:
Do you have pylon viewer installed and have this issue in there?
Perhaps you can check the network optimization tool if you do.

niallodowd, 2021-02-19:
Thanks for the comment! I do have pylon viewer, though I'm not sure where the network optimization tool is, or what it does/how to use it etc.

ridvanozdemir, 2022-09-22:
you can try this code, it worked for me

https://github.com/ridvanozdemir/hardware_trigger_basler_camera