basler/pypylon #334: pip3 install fails after v1.6 on Raspberry Pi4 32-bit with "not a supported wheel on this platform" error.

MarkG-DMI, 2021-02-22:
Apologies in advance if this is covered elsewhere -- I searched and didn't find my specific problem.

Raspberry Pi 4 Model B running Raspberry Pi OS Desktop 20.10 32-bit armv7l.

`root@raspberrypi:/home/pi# uname -a`
`Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux`

**First problem:**  Attempting to install with `pip3 install pypylon` fails:

`root@raspberrypi:/etc# pip3 install pypylon`
`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple`
`Collecting pypylon`
`Could not find a version that satisfies the requirement pypylon (from versions: )`
`No matching distribution found for pypylon`

**Second problem**: Installing directly from any wheel after 1.6  i.e. `https://github.com/basler/pypylon/releases/download/1.7.1/pypylon-1.7.1-cp38-cp38m-manylinux_2_24_armv7l.whl` fails:

`pypylon-1.7.1-cp38-cp38-manylinux_2_24_armv7l.whl is not a supported wheel on this platform.`

Note that installing from `https://github.com/basler/pypylon/releases/download/1.6.0/pypylon-1.6.0-cp37-cp37m-linux_armv7l.whl` works as expected.

`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple`
`Collecting pypylon==1.6.0 from https://github.com/basler/pypylon/releases/download/1.6.0/pypylon-1.6.0-cp37-cp37m-linux_armv7l.whl`
`  Downloading https://github.com/basler/pypylon/releases/download/1.6.0/pypylon-1.6.0-cp37-cp37m-linux_armv7l.whl (7.5MB)`
    `100% |████████████████████████████████| 7.5MB 60kB/s `
`Installing collected packages: pypylon`
`Successfully installed pypylon-1.6.0`

Should I be able to install 1.7.1 in this way?

thiesmoeller, 2021-02-22:
See the remark in 1.7.1 ⚠️ You need at least pip 20.3 to install them.

So update pip first.

python3 -m pip Install pip -U

MarkG-DMI, 2021-02-23:
Fixed!  Having just installed pip I figured it was up to date.  Not so.  Thanks!

closed 2021-02-23 by MarkG-DMI