basler/pypylon #335: Put text / circles ... into LiveImage / Python

pschulze-henke, 2021-02-23:
Hello together,
I would put text / circles ... into the live image. 
With a IDS camera I using   cv2.putText(img, .... and it works.
With a Basler camera I have no idea.
I found this to convert the image:

   // Create an OpenCV image from a pylon image
  openCvImage = cv::Mat(ptrGrabResult->GetHeight(), ptrGrabResult->GetWidth(), CV_8U3, (uint8_t *) pylonImage.GetBuffer());

Unfortunately I am not able to use this.

Can somebody help me with links to examples or other approches?

Thank you verry much
Peter

PolymaTh-EE98, 2021-02-23:
Once you have the image from grabimage (see the sample code) then you can put things on the images using OpenCV or crop Resize etc.

https://github.com/basler/pypylon/tree/master/samples

pschulze-henke, 2021-02-24:
Hi,
thanks for the answer.
Sorry, I have not right explained my issue.
I want to put colored things (e.g. text) into the bw image.

Can somebody help me with links to examples or other approches?

thanks in advance
Peter

closed 2021-02-24 by pschulze-henke