basler/pypylon #337: pypylon fails to import. Windows, version 1.7.1

trevorpeacock, 2021-03-01:
Attempting to import pypylon version 1.7.1 on Windows 10 x64 produces `ImportError: DLL load failed while importing _pylon: The specified module could not be found.`. Version 1.7.0 works fine.

```
(venv) >pip install pypylon
Collecting pypylon
  Using cached pypylon-1.7.1-cp38-cp38-win_amd64.whl (15.3 MB)
Installing collected packages: pypylon
Successfully installed pypylon-1.7.1

(venv) >python
Python 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 23:11:46) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pypylon import pylon
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\...\venv\lib\site-packages\pypylon\pylon.py", line 40, in <module>
    from . import _pylon
ImportError: DLL load failed while importing _pylon: The specified module could not be found.
>>> ^Z

(venv) >pip install pypylon==1.7.0
Collecting pypylon==1.7.0
  Using cached pypylon-1.7.0-cp38-cp38-win_amd64.whl (15.3 MB)
Installing collected packages: pypylon
  Attempting uninstall: pypylon
    Found existing installation: pypylon 1.7.1
    Uninstalling pypylon-1.7.1:
      Successfully uninstalled pypylon-1.7.1
Successfully installed pypylon-1.7.0

(venv) >python
Python 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 23:11:46) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pypylon import pylon
>>> ^Z

(venv) >
```

stefanklug, 2021-03-02:
@trevorpeacock Thanks for your finding. We will look into this.

stefanklug, 2021-03-02:
@trevorpeacock 
Hi, I did a 1.7.2rc1 that should fix the issue. Could you test using `pip install pypylon==1.7.2rc1` ?

trevorpeacock, 2021-03-02:
Imports fine on `1.7.2rc1`. I don't have a camera with me this week to do a full test, I'll report back when I can if there any issues. For now I'd say resolved. Thanks Stefan.

```
(venv) >pip install pypylon==1.7.2rc1
Collecting pypylon==1.7.2rc1
  Downloading pypylon-1.7.2rc1-cp38-cp38-win_amd64.whl (15.3 MB)
     |████████████████████████████████| 15.3 MB 2.2 MB/s
Installing collected packages: pypylon
  Attempting uninstall: pypylon
    Found existing installation: pypylon 1.7.1
    Uninstalling pypylon-1.7.1:
      Successfully uninstalled pypylon-1.7.1
Successfully installed pypylon-1.7.2rc1

(venv) >python
Python 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 23:11:46) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pypylon import pylon
>>> ^Z
```

stefanklug, 2021-03-03:
@trevorpeacock Thanks for the feedback. 
1.7.2 was released to pypi today, so the issue should no longer happen in the wild. Cheers Stefan.

closed 2021-03-03 by stefanklug