basler/pypylon #341: Enum entry is not writable

kostromins, 2021-03-11:
Hi!

I am trying to capture frames from camera triggered by hardware trigger. I configured camera and tested in Pylon Viewer, everything is great and images are acquired on every external hardware trigger. However, when I am connecting to camera from python "Trigger Mode" is being set "Off" and camera enters in continuous grabbing mode without external trigger involved. When I try to set TriggerMode from python code to "Off" it works, however when I try to set it to "On"  I get exception:

  camera.TriggerMode.SetValue("On")
  File "/home/user/miniconda3/envs/test/lib/python3.7/site-packages/pypylon/genicam.py", line 2316, in SetValue
    return _genicam.IEnumeration_SetValue(self, entry)
_genicam.AccessException: Failed to write enumeration value. Enum entry is not writable : AccessException thrown in node 'TriggerMode' while calling 'TriggerMode.FromString()' (file 'Enumeration.cpp', line 137)

I also tried to get list of possible Trigger values and I got interesting output, its like "On" option is missing:

Command: print(camera.TriggerMode.GetSymbolics())
Output: ('Off',)

Any ideas?
Thank you.

PolymaTh-EE98, 2021-03-11:
try
camera.TriggerMode.IntValue = 1

I think that is how I did it.

kostromins, 2021-03-11:
When executing camera.TriggerMode.IntValue = 0 it works, but when executing camera.TriggerMode.IntValue = 1 getting the same error

camera.TriggerMode.IntValue = 1
  File "/home/user/miniconda3/envs/test/lib/python3.7/site-packages/pyp                                                                 ylon/genicam.py", line 2295, in SetIntValue
    return _genicam.IEnumeration_SetIntValue(self, Value, Verify)
_genicam.AccessException: Failed to write enumeration value. Enum entry is not writable : AccessException thrown in node 'TriggerMode' while calling 'TriggerMode.SetIntValue()' (file 'Enumeration.cpp', line 151)

thiesmoeller, 2021-03-11:
TriggerMode is a selected feature.
So you always have to first decide which trigger you want to configure with TriggerSelector

"Trigger Mode | Basler" https://docs.baslerweb.com/trigger-mode

kostromins, 2021-03-12:
I just made couple of another tests and it seems that triggering configuration made via PylonViewer is dropped when opening camera in Python. I even tried to execute pre-saved configuration as UserSet3, by executing:
camera.UserSetSelector = "UserSet3"
camera.UserSetLoad.Execute()
but its seems not affecting Trigger settings at all.

Not sure why trigger settings are not loaded via Userset, but it makes sense why setting TriggerMode throws exception, because default TriggerSelector value is not supprting Trigger "On"

Thanks @thiesmoeller your suggestion works. After setting TriggerSelector = "FrameStart", camera.TriggerMode.GetSymbolics() returns both values as valid and camera.TriggerMode.SetValue("On") is working fine.

closed 2021-03-12 by kostromins

thiesmoeller, 2021-03-12:
The parameters of a camera are only valid while you are in your active
session
Started with Open
Stopped with Close

If you want to persist startup values from pylon viewer store into userset
_and_ configure to use this userset for the next boot

2021, 11:32:

kostromins, 2021-03-12:
I already did it. I stored configuration in userset and pointed it as a default userset. When connecting with Pylon Viewer even after camera is rebooted settings are loaded properly, while as I mentioned trigger settings are not loaded if camera is opened in Python. I tested with other parameters for example PixelFormat. I opened camera in Python, get value to see if PixelFormat is not dropped to default value (it's not), then intentionally changed PixelFormat to some value, read value to see if its changed (it is), and finally read userset in python to see if PixelFormat value goes back to vale stored in userset and it does. So for some reason TriggerType, TriggerMode, TriggerSource and perhaps other related parameters are not loaded from userset when opening camera in Python, so I have to set them manually in code.

thiesmoeller, 2021-03-12:
to be sure that you start from a known configuration,
I typically have these lines in my scripts:

execute "usersetload" on the userset that is configured to be used
on power-up

```
cam.UserSetSelector = cam.UserSetDefault.Value
cam.UserSetLoad.Execute()
```

This will give you a device state, like fresh after power-up

kostromins, 2021-03-12:
Thanks, will try.