basler/pypylon #342: Attribute error:  module 'pypylon' has no attribute 'InstantCamera'

rcbullins, 2021-03-15:
Hello,

I am using a basler camera to implement in deep lab cut live. 

However, when I try to initialize the camera with the following piece of code, "camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())", as is described in the [ README.md](https://github.com/basler/pypylon/blob/master/README.md#installation-from-source) I receive the following attribute error, "module 'pypylon' has no attribute 'InstantCamera'".

Is there a different version of InstantCamera with a different name for pypylon version 1.7.2? Or do I need to add it to my path differently?

Thank you!

thiesmoeller, 2021-03-15:
`from pypylon import pylon`

So you import this way ?

Or are you in the source directory?

rcbullins, 2021-03-15:
I have been importing using 'import pypylon as pylon'.

thiesmoeller, 2021-03-16:
This will just rename pypylon to pylon.
https://docs.python.org/3/reference/simple_stmts.html

Pypylon has two sub modules pylon and genicam

You need the pylon sub module

rcbullins, 2021-03-16:
Okay, it is importing now. Thank you!

closed 2021-03-16 by rcbullins