basler/pypylon #343: Does continuous auto white balance work when grabbing a single image

JustinMBrown, 2021-03-18:
Due to bandwidth limitations I can't stream my cameras, and so I'm using the grabone.py sample as a starting point. But I also need auto white balancing. Does the continuous AWB happen on the camera itself? And if so does it start when camera.Open() is called? Or does AWB require calling camera.StartGrabbing()?

thiesmoeller, 2021-03-20:
To do auto white balancing, the camera has to acquire images ....
And the cameras will only acquire images when you start grabbing ...
So the system behavior is always deterministic.

There might be some work-arounds possible in your case, like reducing the camera bandwidth to some minimal value while "waiting" for the real image. This will throttle down the system, but still keep the AWB running.

JustinMBrown, 2021-03-22:
@thiesmoeller Unfortunately bandwidth is really tight. Could y'all add a new grab command that makes the camera start grabbing frames internally but doesn't send them to the computer? I assume the AWB calculation is done on the camera itself, right?

thiesmoeller, 2021-03-22:
What camera model are you using ?

JustinMBrown, 2021-03-22:
@thiesmoeller  https://www.baslerweb.com/en/products/cameras/area-scan-cameras/ace2/a2a3840-13gcpro/

thiesmoeller, 2021-03-22:
On the Pro models: So even when compression and downscaling (pixel beyond) are on maximum values you run into a bandwidth issue?

JustinMBrown, 2021-03-22:
@thiesmoeller Yes even with all of that and FPS set to 1, I would still get an unstable connection with a few dropped frames. I even followed the Basler guide about making sure jumbo packets are enabled and stuff like that, and made sure we're using an intel pro 1000 ethernet device. It could have been the cables or maybe the switch we were using, but at the point replacing hardware would be a pain, and we don't need the stream anyways. We just need to take a few images at specific times over a period of 10 or so seconds. So a new grab method that makes the camera start grabbing images but doesn't send them to the computer untill GrabOne() is called would be ideal.

A new

thiesmoeller, 2021-03-23:
Independent of your feature request:
What you describe sounds like if the network parameter changes are not set, when you run your test.
E.g. the inter packet delay will reduce the peak bandwidth _massively_  ( which is typically the main root cause at issues like yours) 
```
# Inter-Packet Delay
camera.GevSCPD.SetValue(1000)
```
In pylon viewer you can easily identify a working value and then set this value from your code

thiesmoeller, 2025-12-15:
This question about auto white balance has been addressed in the comments. AWB requires continuous grabbing to work properly. We are closing this issue. For further questions on AWB, please refer to the discussion or open a new issue.