basler/pypylon #345: Failed to load embedded nodemap UsbStreamParams

GaNiziolek, 2021-03-25:
Hello,

I'm having the following problem when run any code in the "Samples" folder.

`Traceback (most recent call last):
  File "/home/gabriel/Documentos/GitHub/Etiquetas-com-API/main.py", line 29, in <module>
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
  File "/home/gabriel/Documentos/GitHub/Etiquetas-com-API/venv/lib/python3.8/site-packages/pypylon/pylon.py", line 1582, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: Failed to load embedded nodemap UsbStreamParams : RuntimeException thrown (file 'PylonUsbStream.h', line 50)`

The same error occurs when a try to connect my camera (Basler acA4024-29uc) in the Pylon Viewer.

I already run the script on `opt/pylon/share/pylon/setup-usb.sh`, but the same error continues.

I'm using Linux Mint 20.1 Cinnamon.

SMA2016a, 2021-05-03:
Are you able to grab images using the Camera emulator in pylon viewer?  Do you start pylon viewer outside of your virtual environment?

would you make a strace log with pylon viewer?

SMA2016a, 2021-05-05:
please try to delete the Genicam cache and try again to open the camera.

/home/user/genicam_xml_cache

closed 2022-06-13 by GaNiziolek