basler/pypylon #346: Camera.StartGrabbingMax loop randomly stops grabbing for 2 minutes then continues

tanovdimitar19, 2021-03-28:
Hello everyone,

below you will the code that I use. It's the test code from the folder samples. I modified it a bit in order to count how many successful and unsuccessful grabs I have from the camera. The code works fine, but every so often the I have unsuccessful grabs for around 2 minutes, and then the code continues working normally. I've checked to see if this happens if I start a live stream from Pylon Viewer, and it doesn't happen. Had you had any similar issues like this. 

P.S. I've ran Bandwidth Manager in Pylon.

*** THE CODE ***

from pypylon import pylon

counter_pass = 0
counter_fail = 0

ip_address = '192.168.1.110'
factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = ptl.CreateDeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = factory.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)

camera.Close()
camera.Open()

while True:

    new_width = camera.Width.GetValue() - camera.Width.GetInc()
    if new_width >= camera.Width.GetMin():
        camera.Width.SetValue(new_width)

    numberOfImagesToGrab = 1
    camera.StartGrabbingMax(numberOfImagesToGrab)
    while camera.IsGrabbing():
        grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

        if grabResult.GrabSucceeded():
            # Access the image data.
            #print("SizeX: ", grabResult.Width)
            #print("SizeY: ", grabResult.Height)
            img = grabResult.Array
            #print("Gray value of first pixel: ", img)
            counter_pass = counter_pass +1

        else:
            counter_fail = counter_fail + 1

        grabResult.Release()

    print ('pass: ' + str(counter_pass) + ' fail: ' + str(counter_fail))

    time.sleep(0)

camera.Close()

SMA2016a, 2021-05-13:
no, I never faced such issues.
Which camera model are you using?

PolymaTh-EE98, 2021-05-13:
I had a similar issue, I used a different model network card and it went away.