basler/pypylon #347: USB connection issues after reboot of system

valli28, 2021-04-06:
I'm using the a2A3840-45umBAS camera and after upgrading my system from a Jetson AGX Xavier to a PC, I have issues when I start my script right after booting. 
It's the same error as [#125 ](https://github.com/basler/pypylon/issues/125), but I can fix it by just plugging out the camera, and simply plugging it in again. 
I'm using a Basler USB3 cable. 
I've tried using a different Basler USB3 cable.
The script is run inside a docker container

I hope someone has an idea as to where the issue resides.

SMA2016a, 2021-05-13:
what is your firmware version?

lars-ek, 2021-05-14:
basler does not survive a reboot, it is not accessible anymore. We had to build a hard reset (cutting power supply for a few seconds).

If there is some kind of software reset, I would be happy :)

valli28, 2021-05-17:
@SMA2016a 
pypylon version: 1.7.2
Pylon version: 6.1.1
Python version: 3.6.9
Device: a2A3840-45umBAS - Basler ace 2 USB3 camera

@lars-ek Actually I didn't have any issues when I was using a Jetson. I only recently migrated to a desktop PC instead, and only then did the issue arise. 

Here is a small exception snippet of how the issue typically arises:
```terminal
    self.cam = pylon.InstantCamera(tl_factory.CreateDevice(dev_info))
  File "/opt/conda/lib/python3.7/site-packages/pypylon/pylon.py", line 1721, in CreateDevice
    return _pylon.TlFactory_CreateDevice(self, *args)
_genicam.RuntimeException: Failed to open device '2676:ba05:2:1:4' for XML file download. Error: 'UX Status: Libusb error: **UNKNOWN**.' : RuntimeException thrown (file 'PylonUsbTL.cpp', line 484)
```

When trying to initialize, it takes around 10-20 seconds for the error to appear. 
If I just unplug and reinsert the USB3 cable, it it as if nothing was ever wrong. 
This is not desirable as this machine will be used every day, so doing this as a part of the boot-up process will result in unnecessary wear and tear.

SMA2016a, 2021-05-17:
<I didn't have any issues when I was using a Jetson. I only recently migrated to a desktop PC instead, and only then did the issue arise.>  The issue seems to be caused by a combination of multiple hardware. e.g camera + USB host controller.
Please try to get a firmware update for your camera first. Contact your local support team to get the firmware update for your camera

valli28, 2021-05-25:
Alrighty, I guess we can close this issue here. Here is a recap:

I contacted my local Basler distributor here in the Nordics and they provided me with a firmware update file for the camera. 
I used Pylon to update the firmware from 1.0.0 -> 2.0.3, which was the version they gave me. 

The issue where the camera is undetected or similar after restarts is now solved, as it no longer happens after this FW update. 

I must admit I find it a bit weird that Basler does not distribute these firmware-packages themselves so that I could have found a solution to my issue sooner, but nevertheless, we found a solution, so thank you @SMA2016a and @lars-ek.

closed 2021-05-25 by valli28