basler/pypylon #348: Pylon displays a black screen on startup

JustinMBrown, 2021-04-06:
When opening Pylon for the first time, it displays a welcome screen that you have to click a few times to get it to close. Well on Ubuntu 18.04, that screen is black and you have to click randomly in order to get it to close. This only happens on some versions of Ubuntu 18.04, so I assuming it's a system library I have to update. 

Also I don't know where else to put this, is there a github or forum for Pylon?

thiesmoeller, 2021-04-06:
https://www.baslerweb.com/en/sales-support/support-contact/#

JustinMBrown, 2021-04-07:
That's pretty lame. What if someone else has the same issue? What if someone already has had the same issue and got it resolved, and now I'm wasting time trying to fix it simply because they didn't ask the question in a public forum?

andreasgau, 2021-04-07:
The information you are looking for is in the pylon release notes (probably an OpenGL related issue, the welcome screen has been developed using QML):
* pylon Viewer: To use the pylon Viewer, you must install the latest Mesa OpenGL driver for
  your system. On Debian-based platforms, you will also need to install the libgl1-mesa-dri
  and libgl1-mesa-glx packages.

andreasgau, 2021-04-07:
Setting the following Qt environment variable may also work:
 export QT_QUICK_BACKEND=software

thiesmoeller, 2025-12-15:
This issue about Pylon Viewer displaying a black screen on Ubuntu 18.04 appears to be related to system libraries or display configuration rather than a pypylon bug. We are closing this issue. For Pylon Viewer issues, please contact Basler support or check system library requirements.