basler/pypylon #350: failed to read memory at 0x200,0x200 bytes. (0xE101800B)

vijay180396, 2021-04-13:
pylon version: `5.0.9.10388 64-bit

when i try connect a gige camera through pylon viewer it show error  **failed to read memory** .Even i try to connect with pypylon source it shown same error.

thiesmoeller, 2021-04-23:
What is your camera model?
Anyhow update to latest pylon as your are using a very old version.

AndresDuque, 2021-05-25:
Hi @thiesmoeller ,

I am working with the CMake and MinGW for cross-compilation. Currently I am getting an error using the Basler API for C++ which is unsupported platform coming from Platfrom.h. It seems the only compilers support you have are from the ABI of each system.

Might I missing a version with support to GCC on Windows?

Could you give me some insights in this topic?

Cheers,
Andres.

AndresDuque, 2021-05-26:
I actually was able to solve my issue using the Pylon C API with CMake. I emailed Basler support in Europe and they were really fast answering with the solution to be able to use MinGW in Windows. Don't hesitate to contact me if you are having the same issue.

vijay180396, 2021-05-27:
@AndresDuque @thiesmoeller  thanks for information

closed 2021-05-27 by vijay180396