basler/pypylon #351: Failed to write memory?

shenjun1994, 2021-04-13:
When  i run the code with some hours，the pylon has raised a error:
Failed to write memory at 0x40224, 0x4 bytes.Timeout,no message received.RuntimeException thrown(file 'gxport.cpp',line 126).
it show the error at camera.ExecuteSoftwareTriggle(),but i use the code along, never meet the error.

vijay180396, 2021-04-17:
me also had same error when i use camera(Gige) more than four hours it through the error

thiesmoeller, 2021-04-23:
This indicates that your network connection is blocking somehow and all GEV retries also failed.

You should check network components . For debugging running wireshark in loop recorder mode in parallel can show what happens in your system

shenjun1994, 2021-04-24:
> This indicates that your network connection is blocking somehow and all GEV retries also failed.
> 
> You should check network components . For debugging running wireshark in loop recorder mode in parallel can show what happens in your system

maybe the version of 1.7.2 is incompatible，when i change the version to 1.6.0, it’s continously running more than one day, but i do not change my code.

thiesmoeller, 2021-05-01:
@shenjun1994 what is your OS and hardware platform?

shenjun1994, 2021-05-05:
> @shenjun1994 what is your OS and hardware platform?
my os is win10 1909, and the pylon version is 6.1, hardwave platform is aca4112-8gc.