basler/pypylon #352: Python freeze on camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

SpangeJ, 2021-04-20:
Hi all,
I have a running software using pypylon and Basler acA USB cameras.
Every now and then Python freezes on the command:
`camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())`

The annoying thing is that the command does not crash, python just freezes indefinitely.
Opening the cameras in Pylon Viewer often result in that it freezes and "goes gray"

Has anybody else encountered similar issues?
Is it possible to add an exception to that command, E.G. a 5-second timeout?

Thank you

thiesmoeller, 2021-04-21:
What is the OS + Architecture, and in case of windows the version of pylon SDK that you use?

SpangeJ, 2021-04-21:
> What is the OS + Architecture, and in case of windows the version of pylon SDK that you use?

NVIDIA Jetson Xavier with JetPack 4.4
GNU/Linux (Ubuntu 18.04)
4.9.140-tegra
aarch64
PyPylon Wheel file: pypylon-1.6.0-cp36-cp36m-linux_aarch64.whl

thiesmoeller, 2021-04-23:
Have your tried to update to latest pypylon 1.7.2

NatanBagrov, 2021-04-23:
Same issue here,
Latest versions (downloaded pylon today, installed `pypylon` via `pip`).
MacOS 11.2.3, Python 3.7.

thiesmoeller, 2021-04-26:
can you provide an `strace` from the hang, to see, where we are blocking ?
Also try to reproduce on our side...

SpangeJ, 2021-05-03:
So I created this script:

`from pypylon import pylon
from time import sleep

while True:
    devices = pylon.TlFactory.GetInstance().EnumerateDevices()
    print('Device enumerate OK')
    GetInstance = pylon.TlFactory.GetInstance().CreateDevice(devices[0])
    print('Create Device OK')
    camera1 = pylon.InstantCamera(GetInstance)
    camera1.Open()
    print('Camera Opened')
    sleep(10)
    camera1.Close()
    print('Camera Closed')
    sleep(50)
`

And ran a strace on it. Please see the output:
[out.txt](https://github.com/basler/pypylon/files/6414469/out.txt)

SMA2016a, 2021-05-05:
Thank you very much for providing the strace log.  I will also try to produce the issue and get back you with the result asap

SpangeJ, 2021-05-05:
Hi,
I have now created a script that is able to detect and timeout the code which freezes.
This is NOT a solution, since the cameras still malfunction (per se - hardware reboot solves the issue), but at least now it is detectable without the software/monitor freezing.

Dependencies: https://pypi.org/project/wrapt-timeout-decorator/

Code:
```
import sys
from pypylon import pylon
from time import sleep
from wrapt_timeout_decorator import timeout, detect_unpickable_objects

@timeout(dec_timeout=None, use_signals=False, timeout_exception=TimeoutError,
         dec_allow_eval=False, dec_hard_timeout=False)
def _test_device():
    print("Starting Test")
    devices = pylon.TlFactory.GetInstance().EnumerateDevices()
    print('Test Passed: Device enumerate')
    GI = pylon.TlFactory.GetInstance().CreateDevice(devices[0])
    print('Test Passed: Create Device')
    camera = pylon.InstantCamera(GI)
    print('Test Passed: Instantiate Device')
    camera.Open()
    print('Test Passed: Camera Opened')
    sleep(1)
    camera.Close()
    print('Test Passed: Camera Closed')

def open_device():
    devices = pylon.TlFactory.GetInstance().EnumerateDevices()
    print('Device enumerate OK')
    GI = pylon.TlFactory.GetInstance().CreateDevice(devices[0])
    print('Create Device OK')
    cam = pylon.InstantCamera(GI)
    cam.Open()
    print('Camera Opened')
    return cam

def close_device(c):
    c.Close()
    print('Camera Closed')

argv = sys.argv
var = float(argv[1]) if len(argv) > 1 else 2
print('INIT')
try:
    _test_device(dec_timeout=var, use_signals=False)
    sleep(.2)
    print('All tests passed!')
except TimeoutError as EXX:
    print(EXX)
    print('Something Wrong with the cameras, cannot connect!')
except TypeError as EXX:
    print(EXX)
    print('Timeout Decorater cannot handle SwigPyObject, the software cannot return anything')
else:
    c = open_device()
    sleep(2)
    close_device(c)
sleep(5)
```

Results on a functioning machine:
```
INIT
Starting Test
Test Passed: Device enumerate
Test Passed: Create Device
Test Passed: Instantiate Device
Test Passed: Camera Opened
Test Passed: Camera Closed
All tests passed!
Device enumerate OK
Create Device OK
Camera Opened
Camera Closed
```

Results on a computer that would freeze:
```
INIT
Starting Test
Test Passed: Device enumerate
Function _test_device timed out after 5.0 seconds
Something Wrong with the cameras, cannot connect!
```

Hopefully, you might use this @NatanBagrov 

@SMA2016a , @thiesmoeller , hopefully, this issue can be resolved in a propper manner.

P.S.
If you try to edit my code to run the timeout decorated function, _test_device(), in a loop. It fails on the second time in the loop.
But my assumption is that this is an issue with the wrapt_timeout_decorator.
Hence: https://github.com/bitranox/wrapt_timeout_decorator/issues/30

SMA2016a, 2021-05-05:
@thiesmoeller @SpangeJ 

I could run my following test code without any problem  on Xavier:
 
L4T 32.1.0 [ JetPack 4.2 ]
   Ubuntu 18.04.2 LTS
   Kernel Version: 4.9.140-tegra
  Python 3.6.9
  pylon ::   1.7.2

Test Code:
from pypylon import pylon
from time import sleep
import os
os.environ["PYLON_CAMEMU"] = "3"

counter=0
while True:
    counter=counter + 1
    devices = pylon.TlFactory.GetInstance().EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RuntimeException("No camera present.")
    else:    
        print('Device enumerate OK')
        GetInstance = pylon.TlFactory.GetInstance().CreateDevice(devices[0])
        print('Create Device OK')
        cam = pylon.InstantCamera(GetInstance)
        print("Using device ", cam.GetDeviceInfo().GetModelName())
        cam.Open()
        print('Camera Opened')
        sleep(10)
        cam.Close()
        print('Camera Closed')
    sleep(10)
    print("next round : "+ str(counter))

Output ::
(pylontest) root@Xaviar1:/wp/pylontest/src# python3 enum_sim.py
Device enumerate OK
Create Device OK
Using device  daA3840-45uc
Camera Opened
Camera Closed
next round : 1
Device enumerate OK
Create Device OK
Using device  daA3840-45uc
Camera Opened
Camera Closed
next round : 2
Device enumerate OK
Create Device OK
Using device  daA3840-45uc
Camera Opened
Camera Closed
next round : 3
Device enumerate OK
Create Device OK
Using device  daA3840-45uc
Camera Opened

SpangeJ, 2021-05-06:
The output of the `OK` Xavier:
```
Device enumerate OK
Create Device OK
Using device  acA1300-200um
Camera Opened
Camera Closed
next round : 1
Device enumerate OK
Create Device OK
Using device  acA1300-200um
Camera Opened
Camera Closed
```

The output of the `Not OK` Xavier:
```
Device enumerate OK
```
I let the software run for 30 minutes, then I did `CTRL + Z`  `sudo kill -9 {PID}`
I.E. the command
`GetInstance = pylon.TlFactory.GetInstance().CreateDevice(devices[0])`
freezes

SMA2016a, 2021-05-06:
@SpangeJ   Meaning that you have got 2 boards and on one pylon working fine but not on the 2ed one?
I did you get the invitation for a remote support session?

SpangeJ, 2021-05-06:
@SMA2016a 
Yes, I got two machines set up next to each other, each with cameras connected to them.
The first machine is Ok, the second started with freeze after some days of testing. If I cycle the power on the second machine the command does not freeze anymore but give it some days and the bug returns.

No, I have not got such an invite.

SMA2016a, 2021-05-07:
Now we understood the issue better.
there are 2 Workarounds:

1) reboot the system.
2) link the genicam cache directory to a directory that is not existing. e.g GENICAM_CACHE_V3_1=/tmp/tmp2

SpangeJ, 2021-05-07:
Thank you @SMA2016a 
I will close the issue for now, since we found a workaround to my problem,
It will be interesting to see if you are able to fix this genicam bug so we don't need a workaround in the future.

closed 2021-05-07 by SpangeJ

namranpan, 2022-09-19:
@SMA2016a  How to execute the second workaround? Is this a linux command linking of directories?

There is no GENICAM_CACHE_V3_1 folder in my system (Nvidia Jetson Xavier). What I can see is a /home/user/genicam_xml_cache. How do I link this to a directory that does not exist?

SpangeJ, 2022-09-29:
`$ export GENICAM_CACHE_V3_1=/tmp/non_existing_folder`