basler/pypylon #353: Streaming via Basler Camera and grab a picture when hardware triggered using python

fatmafarah, 2021-04-20:
Is there a way using Basler Camera daA1280-54um and coding in python, to stream continuously and show it in the screen and grab an image only when the trigger is pressed ?

thiesmoeller, 2021-05-01:
You would normally implement this via IO-Chunks. ( ChunkLineStatusAll)
So you can see if a trigger is pressed during one of the frame starts.
.... 
But the daA1280-54um has no support for IO Chunks. So I would propose to use grab using the background loop and poll the GPIO state in foreground.
And select the current frame to be displayed if you detect a level change on the GPIO.
Depending on your targeted frame rate this would "just work" ( sampling should be possible in the kHz range )

Kamlesh364, 2021-05-22:
> You would normally implement this via IO-Chunks. ( ChunkLineStatusAll)
> So you can see if a trigger is pressed during one of the frame starts.
> ....
> But the daA1280-54um has no support for IO Chunks. [...]

Please! provide me an example code to detect Hardware Triggering via IO-Chunks

Kamlesh364, 2021-05-22:
> Please! provide me an example code to detect Hardware Triggering via IO-Chunks

I just want "True" if the hardware is triggered and "False" otherwise, Please provide me code accordingly..

thiesmoeller, 2021-05-23:
Take a look into our companion repository pypylon-samples https://github.com/basler/pypylon-samples

There is a jupyter notebook that demonstrates chunk handling on he input line ( notebook 4.)

Kamlesh364, 2021-05-23:
> Take a look into our companion repository pypylon-samples https://github.com/basler/pypylon-samples
> 
> There is a jupyter notebook that demonstrates chunk handling on the input line ( notebook 4.)

Thanks for the reference.

thiesmoeller, 2021-05-25:
Example for this use case provided

closed 2021-05-25 by thiesmoeller