basler/pypylon #354: Basler Aces2 USB Cam on Jetson Nano

kotai2003, 2021-04-21:
Hi,
I am trying to install one basler aces2 usb camera on Nvidia's Jetson Nano.
Jetpack version is 4.5.
I would like to know there is someone who have succeeded in installing pylon on Jetson Nano?
Please leave your advices. Thanks.

thiesmoeller, 2021-04-23:
Is your device visible with `lsusb`?

SMA2016a, 2021-04-26:
did you install pylon SDK and  setup the udev rules using pylon setup script? 
It is normaly located in:
 /opt/pylon/share/pylon/setup-usb.sh 

as already asked, you you post the output of lsusb,please?

kotai2003, 2021-05-13:
@SMA2016a 
Sorry for late reply.

As you mentioned, I checked lsusb and re-install the pylon SDK. and I confirm that my basler cam works on Jetson Nano properly.

Thanks so much.

closed 2021-05-13 by kotai2003