basler/pypylon #356: Displaying continuous grabbing from basler camera and show it on Tkinter GUI

fatmafarah, 2021-04-22:
I am trying to feed my gui written with tkinter python with live camera capture with Basler camera USB3. I wrote this code but it's not working. My camera is Basler daA1280-54um (23625361). I want to capture the video and show it in a label or frame, and when there is hardware trigger in the camera take that image and show it in a different label.

`     from pypylon import pylon
      import cv2
      from tkinter import *
      from PIL import Image, ImageTk

      def video_stream():
               tlf = pylon.TlFactory.GetInstance()
               cam = pylon.InstantCamera(tlf.CreateFirstDevice())
              cam.Open()
              cam.UserSetSelector = 'Default'
              cam.UserSetLoad.Execute()
              cam.PixelFormat.SetValue('Mono8')
              cam.Gain.SetValue(17.9)
              cam.ExposureTime.SetValue(60.0)
             cam.TriggerMode.SetValue('Off')
             cam.Gain.SetValue(17.9)
             cam.ExposureTime.SetValue(60.0)

            # Grabing Continusely (video) with minimal delay
          cam.StartGrabbing(1)
          converter = pylon.ImageFormatConverter()

           # converting to opencv bgr format
           converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

            while cam.IsGrabbing():

                grabResult = cam.RetrieveResult(pylon.waitForever, pylon.TimeoutHandling_ThrowException)

                if grabResult.GrabSucceeded():

                         image = converter.Convert(grabResult)
                         b = cam.LineStatus.GetValue()
                         if b==True: #if there is a hardware triggerer, show the image
                            cam.Gain.SetValue(17.9)
                            cam.ExposureTime.SetValue(60.0)
                            img1 = image.GetArray()
                            cv2image = cv2.cvtColor(img1, cv2.COLOR_BGR2RGBA)
                            img = Image.fromarray(cv2image)
                            imgtk = ImageTk.PhotoImage(image=img)
                            lmain2.imgtk = imgtk
                            lmain2.configure(image=imgtk)
                           # Access the image data
                       else:
                             print('im here')
                             cam.Gain.SetValue(17.9)
                             cam.ExposureTime.SetValue(60.0)
                             img1 = image.GetArray()
                             cv2image = cv2.cvtColor(img1, cv2.COLOR_BGR2RGBA)
                            img = Image.fromarray(cv2image)
                            imgtk = ImageTk.PhotoImage(image=img)
                             lmain.imgtk = imgtk
                             lmain.configure(image=imgtk)
                             k = cv2.waitKey(1)
                              if k == 27:
                                   break
                   grabResult.Release()

root=Tk()
app = Frame(root, bg="white")
app.grid()
# Create a label in the frame
lmain1 = Label(app)
lmain2 = Label(app)
lmain1.grid(row=0,column=0)
lmain2.grid(row=0,column=1)
root.mainloop()`

Can someone tell me why it's not working

thiesmoeller, 2021-04-24:
have you checked with : https://github.com/basler/pypylon/issues/72#issuecomment-687710210

for better readability, please edit your issue and remove the single ticks ( these are only for one-line code ) and replace by ` ``` ...code ...``` `
In the edit window of the issues you even have a button for code pasting. just **select all lines of code** and click 
[image]

Technical question: what do you want to reach with ?
`cam.StartGrabbing(1)`

oumaymabelhajsalah96, 2021-06-29:
Hello @fatmafarah , can you help me, I work with a basler camera and tkinter like you.
I can contact you on gmail please.
my gmail is oumayma.belhajsalah96@gmail.com