basler/pypylon #357: CompressionBeyond: provide pypylon sample

ZacDiggum, 2021-04-23:
Hello,

could you please provide a minimal working example of using the CompressionBeyond feature in pypylon?
I found the `ImageDecompressor` class and it's `DecompressImage()` function in pylon.py but I'm not sure how to use it correctly.

thiesmoeller, 2021-04-23:
You should be able to call 

decomp = ImageDecompressor()

res = cam.GrabOne(1000)
py_img = DecompressImage(res)
img = py_img.GetArray()

thiesmoeller, 2021-05-09:
Is this working for you?

thiesmoeller, 2021-05-25:
... we also have an example in our tests
https://github.com/basler/pypylon/blob/e33ed4b235c1ebc188de0151d98859ecd41b743c/tests/pylon_tests/usb/decompressiontest.py

thiesmoeller, 2025-12-15:
This request for a CompressionBeyond sample has been addressed in the comments with usage examples. We are closing this issue. For further questions on CompressionBeyond, please refer to the discussion or open a new issue.