basler/pypylon #358: Installation Fails on Jetson Nano B01

jkerpe, 2021-04-28:
Hi,
I'm having a problem installing pypylon on my Jetson Nano (ARMv8 Processor rev 1 (v8l) × 4). I've already installed pylon successfully but I can't download pypylon with pip3 through github. My Python Version is 3.7.5. No matter which whl-File I select, I always get this error:

pypylon-xxx.whl is not a supported wheel on this platform.

Is this a common error?
Thanks in Advance.

thiesmoeller, 2021-04-29:
It is most possibly the pip version on your system.

See the readme:

(**) The linux binaries are manylinux_2_24 conformant.
This is roughly equivalent to a minimum glibc version >= 2.24.
⚠️ You need at least pip 20.3 to install them.

So update your pip and it should work.

thiesmoeller, 2021-05-25:
User updated pip as described in project readme

closed 2021-05-25 by thiesmoeller