basler/pypylon #360: How do you set the Device User ID for a USB camera in linux?

JustinMBrown, 2021-04-30:
When using a gige camera, you can set the Device User ID as shown in here:

https://docs.baslerweb.com/setting-the-device-user-id

But, that doesn't work for USB cameras. Also, the USB configurator doesn't work on linux.

I tried setting it in Pylon under Device Control > Device User ID, but it doesn't update the name and when I run

`dev_info.GetUserDefinedName()`

it returns the old name.

thiesmoeller, 2021-05-01:
Device user id is a normal feature:

`cam.DeviceUserID = "camera topleft"`

But on USB device user id is part of the Plug'n'Play information descriptors, to be shown during device discovery without opening the device.
The downside of this is, that you have to replug/restart your device to see the changed value in the discovery info.
( if you don't have physical access just calling 
`cam.DeviceReset.Execute()` will restart/replug the device for you...
So you can set it, but it will only appear in the enumeration

thiesmoeller, 2021-05-25:
Did this explanation work for you ?

JustinMBrown, 2021-05-25:
Yes it worked, thank you!

closed 2021-05-25 by JustinMBrown