basler/pypylon #361: PyPylon Executable With Modified Spec File Wont Find Camera On Other PC

teacup-laser, 2021-05-03:
**Problem**
After making an executable for my USB 3 basler camera using the following modification to the spec file:
```
from PyInstaller.utils.hooks import collect_dynamic_libs
    binaries = [
    (lib, ".") for lib, path in
    collect_dynamic_libs("pypylon")
    ],
```
I get an executable that runs just fine on the PC that I used to build on. 

Now I try to run the same executable on a new PC (still using windows 10). When running the executable I get the following error when I try and use the camera:
**error message:**
`_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'tlfactory.cpp', line 694)`

Another notable thing: The green light on the basler camera does not stay on with the new machine. I checked and the port I am using is still USB 3. The light comes on green and turns off after a few seconds. On the build device this does not happen though. 

**What I have tried:**
Using different implementations of the binaries to add the pylon dlls:
```
import pypylon
pypylon_dir = pathlib.Path(pypylon.__file__).parent
pylon_dlls = [(str(dll), '.') for dll in pypylon_dir.glob('*.dll')]
....
binaries=pylon_dlls,
...
```
**The Goal:**
Be able to build the executable on my windows 10 machine and then have anyone also running windows 10 be able to just run the executable.

thiesmoeller, 2021-05-03:
The Basler pylon usb camera drivers are missing, if you just distribute the application alone.

Please check the section "3.2.3 pylon Camera Driver Packages" in the document [Pylon Software deployment guide]( https://www.baslerweb.com/en/sales-support/downloads/document-downloads/pylon-deployment-guide-software-version-6x-installation-guide/)

You can extract msi installer for the drivers from the full SDK install ( as described in the deployment guide ).

For your application you than wrap this into a small installer ( e.g. by using NSIS or WIX) , that installs the driver msi and your application onto the target machine.

This is specific to windows. On Linux and MacOS we are using generic driver architecture that is always available.

thiesmoeller, 2025-12-15:
This issue about PyInstaller executables not finding cameras has been addressed with workarounds for including DLLs. We are closing this issue. For further questions on building executables, please refer to the discussion or open a new issue.