basler/pypylon #363: Grab for 4 Gige Cameras Issues

cvixxt, 2021-05-11:
Hi everyone, I want to ask some questions.
I need to know how to grab for 4 Gige cameras with multi thread in pypylon.
But I don't have new ideas to develop with pypylon.
I'm using the pypylon sample code "opencv.py & grabmultiplecameras.py" to develop, I catch the 4 cameras and it can work, but I am running the code, Gige Cameras after about 6-8 hours of execution, the camera will jump out of the loop for no reason and the execution ends. And I need to know how can I use multi thread to grab for 4 Gige cameras.

Q1. Why the Gige Cameras will jump out of the loop for no reason and the execution ends after about 6-8 hours of execution?

Q2. I need to know how to grab for 4 Gige cameras with multi thread in pypylon.

If anyone wants to know how did I development, I can provide the code. Thanks.

SMA2016a, 2021-05-13:
1) actually, there is no reason for this issue, unless there is some bandwidth issue. 
       *do you use any network switch?
       *or 4 Port NIC?
       * try to increase the value for the interpacket delay
2)  yes, post your code.

cvixxt, 2021-05-17:
Hi SMA2016a ,
Thank you for answer the question.
1. We didn't use any network switch and 4 port NIC. 
2. OK, I will upload it.

The code is below this line
---------------------------------

import os

os.environ["PYLON_CAMEMU"] = "3"

from pypylon import genicam
from pypylon import pylon
import cv2
import sys
from multiprocessing import Process

maxCamerasToUse = 3   #Modify the number of cameras, enter 5 if there are 5
exitCode = 0
countOfImagesToGrab = 1000000000
countImage =0

def grabOneCam():

    try:
        # Get the transport layer factory.
        tlFactory = pylon.TlFactory.GetInstance()

        # Get all attached devices and exit application if no device is found.
        devices = tlFactory.EnumerateDevices()
        if len(devices) == 0:
            raise pylon.RuntimeException("No camera present.")

        # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
        cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))

        l = cameras.GetSize()

        # Create and attach all Pylon Devices.
        
        for i, cam in enumerate(cameras):
            cam.Attach(tlFactory.CreateDevice(devices[i]))

            # Print the model name of the camera.
            print("Using device ", cam.GetDeviceInfo().GetModelName())

        # Starts grabbing for all cameras starting with index 0. The grabbing
        # is started for one camera after the other. That's why the images of all
        # cameras are not taken at the same time.
        # However, a hardware trigger setup can be used to cause all cameras to grab images synchronously.
        # According to their default configuration, the cameras are
        # set up for free-running continuous acquisition.
        cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
        converter = pylon.ImageFormatConverter()
        # Grab c_countOfImagesToGrab from the cameras.
        while cameras.IsGrabbing():
        #for i in range(countOfImagesToGrab):
            if not cameras.IsGrabbing():
                break
            grabResult = cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

            # When the cameras in the array are created the camera context value
            # is set to the index of the camera in the array.
            # The camera context is a user settable value.
            # This value is attached to each grab result and can be used
            # to determine the camera that produced the grab result.
            cameraContextValue = grabResult.GetCameraContext()

            # Print the index and the model name of the camera.
            print("Camera ", cameraContextValue, ": ", cameras[cameraContextValue].GetDeviceInfo().GetModelName())

            #To determine the name of the current camera, you need to determine by yourself which window the screen show is in
        
            if cameraContextValue ==0:

                image = converter.Convert(grabResult)
                img = image.GetArray()
                cv2.namedWindow('title', cv2.WINDOW_NORMAL)
                cv2.imshow('title', img)
                k = cv2.waitKey(1)
                if k == 27:
                    break
            
            if cameraContextValue == 1:
                image = converter.Convert(grabResult)
                img = image.GetArray()
                cv2.namedWindow('title1', cv2.WINDOW_NORMAL)
                cv2.imshow('title1', img)
                k = cv2.waitKey(1)
                if k == 27:
                    break
            if cameraContextValue==2:
                image = converter.Convert(grabResult)
                img = image.GetArray()
                cv2.namedWindow('title2', cv2.WINDOW_NORMAL)
                cv2.imshow('title2', img)
                k = cv2.waitKey(1)
                if k == 27:
                    break
            if cameraContextValue==3:
                image = converter.Convert(grabResult)
                img = image.GetArray()
                cv2.namedWindow('title3', cv2.WINDOW_NORMAL)
                cv2.imshow('title3', img)
                k = cv2.waitKey(1)
                if k == 27:
                    break

            #Press ESC to stop capturing

        
            # Now, the image data can be processed.
            print("GrabSucceeded: ", grabResult.GrabSucceeded())
            print("SizeX: ", grabResult.GetWidth())
            print("SizeY: ", grabResult.GetHeight())
            #img = grabResult.GetArray()
            #print("Gray value of first pixel: ", img[0, 0])
            if grabResult.GrabSucceeded():
                grabResult.Release()
            # Access the image data
        

    finally:
        return
    #except genicam.GenericException as e:
        # Error handling
        #print("An exception occurred.", e.GetDescription())
        #exitCode = 1

    # Comment the following two lines to disable waiting on exit.
    #sys.exit(exitCode)

if __name__ == '__main__':
      grabOneCam()
      print("Finished")

SMA2016a, 2021-05-17:
@cvixxt thank you very much for posting the code.

is there any reason why you do not print the error description? this may give you any hint on the issue.

#print("An exception occurred.", e.GetDescription())

There is certainly nothing wrong with your code. 

Would you share some information on your operating system,please?  Windows or Linux?   Pylon version , camera model etc..

cvixxt, 2021-05-17:
Hi  @SMA2016a ,
I only want to shoot 4 cameras in a row, and I don't want to interrupt the loop due to camera errors, so I take out the issus for this sample.
My operating system is Windows 10, Pypylon version is the github the newest version, camera model is all of the acA2500-14gc.
Thank you.

SMA2016a, 2021-05-17:
@cvixxt   Then change pylon.TimeoutHandling_ThrowException  to pylon.TimeoutHandling_Return.  In case of timeout the grabresult will be NULL.  You need to check  it before you use the grabresult

cvixxt, 2021-05-18:
Hi @SMA2016a ,
Thank you for answer the question.
I will test your suggestion before replying to you.

thiesmoeller, 2025-12-15:
This issue about cameras jumping out of loop after extended operation requires more information about error handling, network configuration, and system logs. Since we haven't received updates, we are closing this issue. If you are still experiencing this problem, please open a new issue with detailed error logs and network diagnostics.