basler/pypylon #364: how to control the basler's back light

kotai2003, 2021-05-13:
Hi Folks,

I am trying to use the backlighting for application.
The backlighting was fabricated by basler. I could control this lighting with pylon viewer.
The manual also told that pylon SDK support this lighting.

What I would like to know 
1. How to control this lighting with **pyplon** at python.
2. Any small information will be welcome. (command, class information.)

[https://www.baslerweb.com/jp/products/vision-components/basler-lighting/basler-camera-light-back-120x120-white/](
)
[image]

SMA2016a, 2021-05-13:
Do you have a camera that supports controlling light controller?
you can check first with pylon viewer if the light controller can be seen in the cameras parameters list.

or simply post the camera model and the firmware version,please

kotai2003, 2021-05-13:
@SMA2016a Thanks for your message.
I have a camera which model number is  **Basler acA1920-40uc**.  
I could see the light controller can be seen on pylon viewer. 

_Captured image of pylon viewer_
[image]

_The backlighting was controlled with pylon viewer._
[image]

thiesmoeller, 2021-05-13:
@kotai2003 , every feature that you see in pylon viewer, is also available in pypylon automatically.
To learn how this works please check https://github.com/basler/pypylon-samples and the referenced webinar.

So please look in the "Feature Documentation" pane of pylon viewer for the feature names.
Or look up docs.baslerweb.com for your camera and the light control features

kotai2003, 2021-05-13:
@thiesmoeller Thanks for your comment.

I found the C++ code for back light control. 

Could I find the corresponding code of this function at Python language  at pypylon-samples page?

`
<html><head><meta http-equiv="Content-Type" content="text/html; charset=utf-8" /><style type="text/css">
p, li { white-space: pre-wrap; }
</style></head><body>
<h3 style=" margin-top:14px; margin-bottom:12px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><span style=" font-size:large; font-weight:600;">Light Device Operation Mode</span></h3>
<p style=" margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><span style=" font-weight:600;">Sets the operation mode of the light device.</span><br />Sets the operation mode of the light device. <br /><br /><a href="https://docs.baslerweb.com/?utm_campaign=pylon&amp;utm_source=6.1.1.18345&amp;utm_medium=pylonviewer&amp;rhcsh=1&amp;filter=Camera:acA1920-40uc&amp;rhmapid=BslLightDeviceOperationMode"><span style=" text-decoration: underline; color:#0000ff;">Get Help</span></a><br /></p>

  | Node Name: |   | BslLightDeviceOperationMode
-- | -- | -- | --
  | Interface Type: |   | IEnumeration
  | C++ Enum Type: |   | BslLightDeviceOperationModeEnums
  |   |   |  
  | C++ sample code: |   | camera.BslLightDeviceOperationMode.SetValue(BslLightDeviceOperationMode_On);BslLightDeviceOperationModeEnums e = camera.BslLightDeviceOperationMode.GetValue();

<p style=" margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><br /><br /></p></body></html>`

thiesmoeller, 2021-05-13:
In pypylon-samples there are other samples, which demonstrate that from c++ to python is mostly search and replace.

This is your code above in python

```python
camera.BslLightDeviceOperationMode.SetValue("On")

e = camera.BslLightDeviceOperationMode.GetValue()
```

kotai2003, 2021-05-15:
@thiesmoeller 

I tried your code with my IDE. 

```
from pypylon import pylon
import numpy as np

from pypylon import pylon
import cv2

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

camera.BslLightControlMode.SetValue("ON")
```
and this is the result.
The IDE also indicated that camera instance can not find the declaration (**BslLightControlMode**) to go with 
```
camera.BslLightControlMode.SetValue("ON")
```

```
Traceback (most recent call last):
  File "C:/Users/seong/CloudStation/AI_development/124.pypylon/develop_code/00.backlight_control.py", line 9, in <module>
    camera.BslLightControlMode.SetValue("ON")
  File "C:\Users\seong\Anaconda3\envs\keras-ocr-ppang\lib\site-packages\pypylon\genicam.py", line 2316, in SetValue
    return _genicam.IEnumeration_SetValue(self, entry)
_genicam.InvalidArgumentException: Feature 'BslLightControlMode' : cannot convert value 'ON', the value is invalid. : InvalidArgumentException thrown in node 'BslLightControlMode' while calling 'BslLightControlMode.FromString()' (file 'enumeration.cpp', line 134)

```

thiesmoeller, 2021-05-15:
On ... Not ON

kotai2003, 2021-05-15:
@thiesmoeller 
Sorry for my mistake. 
Finally, I could control this back light with python. Thank you so much for your advice.

This is the fully working code.

```
from pypylon import pylon

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

#Light Control Mode
camera.BslLightControlMode.SetValue('On') # Off or On
e1 = camera.BslLightControlMode.GetValue()
print('Bslight Control Mode: ', e1)

#Light Device Brigtness
camera.BslLightDeviceBrightness.SetValue(20) # range between 0 to 100
d = camera.BslLightDeviceBrightness.GetValue()
print('Bslight Brightness [0.0-100.0]: ', d)

#Light Device Operation Mode
camera.BslLightDeviceOperationMode.SetValue('Off') # Off , On , Strobe
e2 = camera.BslLightDeviceOperationMode.GetValue()
print('Bslight Operation Mode: ', e2)

```

closed 2021-05-18 by thiesmoeller