basler/pypylon #365: Basler pylon software trigger not working.

TomsRom, 2021-05-17:
Hello,
I'm trying to trigger Basler Aca1440-220uc with software trigger, but nothing is happening. My future plan is use hardware trigger on Jetson Nano through GPIO pins. For start I wanted to try software trigger to see if it will work, but when I press Software Signal Pulse button nothing happens. I tried camera on Windows 10 and Ubuntu on Jetson Nano. 
I added picture with my acquisition control configuration.
[image]

thiesmoeller, 2021-05-17:
https://docs.baslerweb.com/trigger-software.html

"Software Signal 1"  != "Software"

thiesmoeller, 2025-12-15:
This issue about software trigger not working has been addressed in the comments with configuration guidance. We are closing this issue. For further questions on software triggers, please refer to the discussion or open a new issue.