basler/pypylon #367: Unable to load embedded nodemap when running in docker container with restart policy.

Lochrome, 2021-05-25:
When I first launch my pypylon application in a docker container with a restart policy set to always, everything works fine. However, when I reboot my device and my container is restarted, when I try to open a connection to my camera I run into the following error: "Failed to load embedded nodemap UsbStreamParams". 

Some people have run into this issue in non-docker related contexts and suggested that the answer is to delete the following directory: /home/user/genicam_xml_cache. This does not seem to be the solution to my issue because when I restart my docker container that directory does not exist.

Docker Version: 19.03.8-ce
Pylon Version: 6.1.3.20159
PyPylon Version: 1.7.2

Lochrome, 2021-06-11:
BTW:

Docker OS: arm64v8 Ubuntu Xenial

Any suggestions on how to resolve this issue?

andreasgau, 2021-06-14:
Please try the following, it will disable the GenICam cache:
Link the genicam cache directory to a directory that is not existing. e.g GENICAM_CACHE_V3_1=/tmp/tmp2

The error could be related to the GenICam cache. Especially if you use more than one USB camera in multiple threads or processes.

Anyway it would be very interesting, if the error does or does not occur anymore. This will help to find the cause of the problem.

Lochrome, 2021-06-14:
Thank you for your help! This resolved the issue for me!

thiesmoeller, 2025-12-15:
This issue about nodemap loading in Docker appears to be related to Docker container state management rather than a pypylon bug. We are closing this issue. If you believe this is a pypylon specific bug, please open a new issue with detailed Docker configuration and reproduction steps.

closed 2025-12-15 by thiesmoeller