basler/pypylon #372: GRAB AND SAVE IMAGES USIUNG MULTIPLE CAMERAS AT SAME TIME

thasnimolvs, 2021-06-07:
HEREBY ATTACHING A CODE WITH ERROR , OF INDEX . HEREBY ATTACHING MY CODE, HELP ME TOP FIND RESULT
###############################################################################
from pypylon import pylon
from pypylon import genicam
from datetime import datetime

import time

# Import OpenCV
import cv2
from PIL import Image

# Import os for creating directories.
import os

#os.environ["PYLON_CAMEMU"] = "2"
import sys
# Import shutil for deleting folders.

import shutil

# Import threading for creating threads.
import threading

import numpy as np
outputDir = '/mnt/ramdisk/output'
#outputDir = 'output'

###############################################################################
# Global variables
###############################################################################
  
countOfImagesToGrab = 200 # Number of images to be grabbed
maxCamerasToUse = 3# Limits the amount of cameras used for grabbing
#timeout = 5000 #max time to wait for grabbed image
#serial_numbers = ['40092327', '40092332']
serial_numbers = ['40092327', '40092332','40083731']
# The exit code of the sample application.
#exitCode = 0
    
tlFactory = pylon.TlFactory.GetInstance()

# get all attached devices
devices = tlFactory.EnumerateDevices()

# look for unique devices that match the serial numbers
camera_devices = []
for device in devices:
    serial = device.GetSerialNumber()
    
    for sn in serial_numbers:
        if serial == sn:
            camera_devices.append(device)   
            serial_numbers.remove(serial)
            break                        
            
camera_devices = tuple(camera_devices)
 
 #camera_devices = tuple(camera_devices)

 # exit if no camera found
if len(camera_devices) == 0:
    raise pylon.RuntimeException("No camera present.")
    
print(str(len(camera_devices)) + ' cameras found.')

# Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
cam_array = pylon.InstantCameraArray(min(len(camera_devices), maxCamerasToUse))

    # set camera parameters
# camera.MaxNumBuffer = 5200 #count of buffers allocated for grabbing

class CameraImage:
    
    def __init__(self, device, camera_devices= []):
        self.runningStatus = False
        self.thread = None
        self.imageList = []
        #self.devices
        self.device = device
        #self.camera_devices = []
        self.camera_devices= []
       # self.cam_array=cam_array
        
    def frameGrabber(self):

        #cam_array = pylon.InstantCameraArray(min(len(camera_devices), maxCamerasToUse))
        for i, camera in enumerate(cam_array):
            
        # Create an instant camera object
            camera.Attach(tlFactory.CreateDevice(self.camera_devices[i]))
    
        #camera name
            #camera_name = '_'.join([camera.GetDeviceInfo().GetModelName(),
                                    #camera.GetDeviceInfo().GetSerialNumber()])
        
            camera.Open()
        

            camera.TriggerSelector.SetValue("FrameBurstStart")
            camera.TriggerMode.SetValue("On") #hardware trigger
            camera.TriggerSource.SetValue("Line4")
            camera.TriggerActivation.SetValue('RisingEdge')
            #camera.StartGrabbingMax(countOfImagesToGrab)
            #camera.ExposureAuto.SetValue('Off')
            #camera.ExposureTime.SetValue(200)
            #cam.ExposureTime = 3000
            camera.Gain.SetValue(10)

            camera.PgiMode.SetValue("On")
            #// Enable the PGI feature set (color cameras only)
            #camera.DemosaicingMode.SetValue(DemosaicingMode_BaslerPGI)
            #// Configure noise reduction (if available)
            camera.NoiseReduction.SetValue(2.0)
            #// Configure improved sharpness (if available)
            camera.SharpnessEnhancement.SetValue(3.98437)
            #camera.PgiMode.SetValue(PgiMode_On)
            #camera.NoiseReduction.SetValue(1.5)
            camera.AcquisitionFrameRateEnable.SetValue(True)
            camera.AcquisitionFrameRate.SetValue(200.0) 
            camera.Width= 1440
            camera.Height=1080
            camera.ExposureTime = 3000
        
           
            startTime = time.time()
            camera.StartGrabbingMax(countOfImagesToGrab)
            while camera.IsGrabbing():
                grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

                # Image grabbed successfully?
                if grabResult.GrabSucceeded():
                    serial_number = '_'.join([camera.GetDeviceInfo().GetModelName(),
                                camera.GetDeviceInfo().GetSerialNumber()])
                    print('Serial No:%s' %serial_number) 
                    imageDict = {}
                    imageDict['filename'] = datetime.now().strftime("%Y%m%d-%H%M%S%f-{}.tiff".format(serial_number))
                    imageDict['image'] = grabResult.GetArray()
                else:
                     print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
                grabResult.Release() 
            # camera.Close() 
            
            
                self.runningStatus = True
                # Push the frame to the queue.
                self.imageList.append(imageDict)

            self.runningStatus = False
            endTime = time.time()
            print('Total time = %f' % (endTime - startTime))

    def startGrabbing(self):
            self.thread = threading.Thread(target=self.frameGrabber)
            self.thread.start()
    ###############################################################################
# Class to save video frames.

class SaveImage:

    def __init__(self, cameraImage):
        self.cameraImage = cameraImage
        self.thread = None
    
    def imageSaver(self):
        totalTime = 0
        while not self.cameraImage.runningStatus:
            time.sleep(0.001)
        while True:
            if not self.cameraImage.runningStatus and len(self.cameraImage.imageList) == 0:
                break
            if len(self.cameraImage.imageList) == 0:
                continue
            else:
                imageDict = self.cameraImage.imageList.pop(0)
                img = imageDict['image'] 
                #print('img')
                filename = imageDict['filename']
                #print('filename')
                startTime = time.time()
                #cv2.imwrite('%s/%s' % (outputDir, filename), img)
                im = Image.fromarray(img)
                #print('im')
                im.save('%s/%s' % (outputDir, filename))
                totalTime = totalTime + (time.time() - startTime)
        print('Total time for saving: %f' % totalTime)
                 
    def startSaving(self):
        self.thread = threading.Thread(target=self.imageSaver)
        self.thread.start()
###############################################################################
# Main code
###############################################################################

   
###############################################################################
# Main code
###############################################################################

# Reset the output directory.
if os.path.isdir(outputDir):
    shutil.rmtree(outputDir)
os.mkdir(outputDir)

startTime = time.time()
# Create the video object.
for i, camera_devices in enumerate(devices):
    cameraImageObj = CameraImage(device)
    saveImageObj = SaveImage(cameraImageObj)
    cameraImageObj.startGrabbing()
    saveImageObj.startSaving()
    cameraImageObj.thread.join()
    saveImageObj.thread.join()

# End timing.
endTime = time.time()

# Print the time taken.
print("Total time taken = %f seconds." % (endTime - startTime))

###############################################################################
# End of file.
###############################################################################

###############################################################################
# End of file.
###############################################################################

        
#try:
    
   # tlFactory = pylon.TlFactory.GetInstance()
    
    # get all attached devices
   # devices = tlFactory.EnumerateDevices()
    
    # look for unique devices that match the serial numbers
   
  

        # Print the model name of the camera.
       # print("Setup of camera ", camera_name, " complete.")
        
#except genicam.GenericException as e:
    # Error handling.
   # print("An exception occurred.")
    #print(e.GetDescription())
    #exitCode = 1        
    
#try:  
   
#except genicam.GenericException as e:
    # Error handling.
  #  print("An exception occurred.")
  #  print(e.GetDescription())
  #  exitCode = 1

#sys.exit(exitCode)

thiesmoeller, 2021-06-07:
Hi @thasnimolvs 
Please edit and format your issue according to [formatting code in GitHub docs](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) to make it readable.

Plus provide
Operating system
Camera model
Pypylon version
And the output of calling your script

thasnimolvs, 2021-06-07:
Camera model is basler ace aca 1440-220 um   
Using ubuntu 18.04, 
Pylon software

thasnimolvs, 2021-06-08:
[pgm.zip](https://github.com/basler/pypylon/files/6614117/pgm.zip)
hi @thiesmoeller

thiesmoeller, 2021-06-08:
@thasnimolvs  ... when "HEREBY ATTACHING A CODE WITH ERROR , OF INDEX " is the issue you report. please show what is not working, by posting e.g. the error message.

To format code see above:  [formatting code in GitHub docs](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks)

thasnimolvs, 2021-06-11:
@thiesmoeller
Sir,
in this code whenever i tried
[image]
this is the error message i got

thasnimolvs, 2021-06-11:
@thiesmoeller 
https://github.com/basler/pypylon/issues/371
here i posted one more code related to this but it is working properly , but shows an error like 
[image]
[image]
[image]

and i found one solution in github so based on that i changed the bandwidth but same error again
after that i gave 3 different trigger signal to 3 cameras 
like 1st start by 0,10, 20,
2nd 3,13,23
3rd 6,16,26
again same error 
[image]
System Details:
UBUNTU 18.04
PYPYLON 1.6
PYLON 6.2.0
ramdisk size is 1gb

thiesmoeller, 2021-06-12:
first: the camera is still open by some other process. Perhaps in your IDE the background threads are somewhere still running.
         so check this. the camera can only be open once ( like the error message reports )

second: have you followed the steps to increase your transfer size? and increase the usbfs memory?
        as described in the readme, when you install the pylon for linux package on your system?
        other reason could also be very bad cables or connectors, so the number of failed USB transfers is way out of normal levels

thasnimolvs, 2021-06-12:
@thiesmoeller 
second:
sir, in your that github solution related to bandwidth manager , i tried it but no change, i dont have any awareness about usbfs . 
no sir,  didnt do  anything related to usbfs and all - till now  , am not aware of this things sir, i will search for it. do you have any links related to increasing transfer size  and installing pylon packages on the linux based system?

thasnimolvs, 2021-06-15:
@thiesmoeller 
sir, when i increased the maximum number of buffers , it is  working ,
thankyou. 
 but i have one more doubt, with this actually 3 cameras working at same time with 200fps, but for 3 cam it tooks more time. why it is like that?
[image]

thiesmoeller, 2021-06-17:
If you are implementing code, that is running at framerate, you should measure the time you need for the processing.
Propose to log this over longer times, to identify if there are events, where e.g. saving an image takes longer.

In your example you only have 5ms per frame to store the data to the filesystem.
The number of buffers, that you queue in per camera allows to compensate for fluctuations that happen when saving your image data ( > 5ms )
But of course this should only occur seldom.

In a perfect world you only need two buffers:
buffer-0 -> the buffer where the current live video data from the camera is stored into
buffer-1 -> the buffer that is currently written to a file

If saving is _always_ faster than 5ms buffer-1 will be added to the queue, before it is needed by the capture engine

thasnimolvs, 2021-06-21:
@thiesmoeller 
hi,
i have one more doubt, when i run this code i set the count of images to 4000,
fps is 200 , and at hardware triggering for every 5ms set the frequency as 200hz,
but it saves 12000   images from total 3 cameras,, but noramally fps means it will save thiese many no of images in 1 sec , here 200 images at each sec from 3 cameras, toral 600 images , why it will save 12000

2. if i want to run the code for 10minut or 5 minut which value should i need to change,
    a, fps, count of images 
(also basler camera capacity maxm 227 fps )

could you please answer these question too, for my task

thasnimolvs, 2021-06-23:
@thiesmoeller 
Sir, I want to run this code for 10 minutes , currently what am doing is increasing the count of images to 16k something , and set the fps to 60, 
But without setting the count....it automatically capture images to folder for 10 minut..how it's possible with this code ..eg : as shown in figure..it will took 534 sec.ie 8.9 minut. .. 
[image]

thiesmoeller, 2021-06-23:
If your hardware trigger is 200 Hz and your camera is capable of more than 200fps you are fine on this side.
You _have_ to make sure that you are capable to process and save your images in less than 1/200s on average.

If you want to run in your code for 10 minutes ... you have to grab 200 * 60 * 10 frames.

If your number of images is not matching your expectation, one thing to focus on is, if the trigger is not operating at 200fps
or if you loose images in between...

Your code could be cleaned up. If you are fast enough, you could stay in the camera thread to handle the saving and thus save some copying of the video data ..

thasnimolvs, 2021-06-26:
@thiesmoeller 
sir,
actually in my case for 60fps with 3 camera,
60*10*60*3=10,8000 these no of images will save to the folder , but using this code if i use ramdisk or normal disk it will take more time some times doubles, 
so is it possible to reduce that timing?
what is the reason behind it?
becoz when i use single cam, it will save exact no of images with correct timing.
in one comment you mentioned about the  buffers,
but i dont get much idea behind it?
could you plz give any suggestion for reducing the time for capture?

thasnimolvs, 2021-07-05:
@thiesmoeller 

helo sir,
one more doubt, at 20 fps to 80 fps for 10 minutes it capture and save images  properly, but after that it shows one error,, when i run 200fps it shows the error as given below
[image]

thiesmoeller, 2021-07-05:
As I don't know your code ... 
But could it be that you are using some formula to.calculate the number of images and the result is a float?
Please cast to an `int`  before giving the value to StartGrabbingMax

thasnimolvs, 2021-07-05:
@thiesmoeller 
No sir I directly calculate no of images and given to it 
##############################################################################
# Imports
###############################################################################
from pypylon import pylon
from pypylon import genicam
from datetime import datetime

import time

# Import OpenCV
import cv2
from PIL import Image

# Import os for creating directories.
import os

#os.environ["PYLON_CAMEMU"] = "2"
import sys
# Import shutil for deleting folders.
import shutil

# Import threading for creating threads.
import threading

import numpy as np

###############################################################################
# Parameters
###############################################################################
countOfImagesToGrab=4000

#outputDir = '/mnt/ramdisk/output'
outputDir = 'output'

###############################################################################
# Global variables
###############################################################################
  # Get the transport layer factory.
tlFactory = pylon.TlFactory.GetInstance()
# Get all attached devices and exit application if no device is found.
devices = tlFactory.EnumerateDevices()
print(devices)

class CameraImage:
    
    def __init__(self, device):
        self.runningStatus = False
        self.thread = None
        self.imageList = []
        self.device = device
        
    def frameGrabber(self):
        cam = pylon.InstantCamera(tlFactory.CreateDevice(self.device))
        
        cam.Open()
        cam.TriggerSelector.SetValue("FrameStart") # FrameBurstStart is used for capturing a series of images with a single signal, please choose FrameStart if wish to have one signal one image
        #cam.TriggerDelay.SetValue(300)
        cam.TriggerSource.SetValue("Line4")
        cam.TriggerMode.SetValue("On")
        cam.TriggerActivation.SetValue('RisingEdge')
       # // Select the frame start trigger
      #  cam.TriggerSelector.SetValue(TriggerSelector_FrameStart);
       # // Set the delay for the frame start trigger to 300 µs
        
        #camera.Gain.SetValue(22.5)
        #camera.Gain.SetValue(23.5)
        cam.PgiMode.SetValue("On")
        cam.NoiseReduction.SetValue(2.0)
[... 99 lines cut ...]
cameraImage1.startGrabbing()
saveImage1.startSaving()
cameraImage2.startGrabbing()
saveImage2.startSaving()
cameraImage3.startGrabbing()
saveImage3.startSaving()

cameraImage1.thread.join()
saveImage1.thread.join()
cameraImage2.thread.join()
saveImage2.thread.join()
cameraImage3.thread.join()
saveImage3.thread.join()

# End timing.
endTime = time.time()

# Print the time taken.
print("Total time taken = %f seconds." % (endTime - startTime))

###############################################################################
# End of file.
############################################################
This is my code..but upto 80 fps it's working properly.
If fps is 200 , no of img= 200* 60*10* 3 like

thasnimolvs, 2021-07-05:
is it due to any bandwidth issue ?
becoz i alrdy increase the buffer size and all
@thiesmoeller 
then by direct calcualtion put the no of images in code

thiesmoeller, 2021-07-05:
Repeating from before
> If your hardware trigger is 200 Hz and your camera is capable of more than 200fps you are fine on this side.
> You _have_ to make sure that you are capable to process and save your images in less than [...]

Have you measured the performance of your code?
Are you capable to process frames faster than 1/200 s?
The usage of a thread to save and PIL is not very efficient as there are image copies involved.
Saving with opencv from the grab loop is faster.

And check your hardware trigger source if it is giving stable 200hz

thasnimolvs, 2021-07-10:
@thiesmoeller 
OK sir checked

is it possible to run the code continuously, until that variable is true , then when i set it to false  then it will terminate ?
 without giving time and count of images?
i tried to add awhile loop inside that 1st class but it is not working!!!!!!!

thasnimolvs, 2021-07-11:
@thiesmoeller 
HI sir 
is it possible to run the code without using that cam.startgrabbingmax parameter?
becoz,i want to work it as in infinite loop