basler/pypylon #373: Failed to load embedded nodemap UsbStreamParams using PyPylon and gRPC on Linux

meredithheskethfortescue, 2021-06-07:
I want to setup a gRPC server to use a camera (Basler acA1920) with PyPylon. 
While it all works fine on a Windows 10 Desktop, the two packages seem to be incompatible on a Raspberry Pi (Raspbian Buster).

Both work fine on their own. I can grab images with the pylon viewer and via python scripts.
I've already run `setup-usb.sh`.

## First minimal example
Importing pypylon and grpc is enough to let them crash with a very short error message.

```python
from pypylon import pylon
import grpc
```
```
free(): invalid pointer
```

## Second minimal example
Importing pypylon first and grpc second works for the first moment. But it crashes as soon as a TransportLayer is created.
But now we get a more verbose error message.

```python
import grpc
from pypylon import pylon

tl = pylon.TlFactory.GetInstance()
tl.CreateFirstDevice()
```
```
Traceback (most recent call last):
  File "minimal_example_getinstance.py", line 7, in <module>
    tl.CreateFirstDevice()
  File "/root/camera_addon/venv/lib/python3.8/site-packages/pypylon/pylon.py", line 1582, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: Failed to load embedded nodemap UsbStreamParams : RuntimeException thrown (file 'PylonUsbS                                                                                                                        tream.h', line
```

thiesmoeller, 2021-06-08:
which python version ?
I can run both without any issues on 3.9.5

meredithheskethfortescue, 2021-06-14:
> which python version ?
> I can run both without any issues on 3.9.5

This is my current setup but I will try it with 3.9.5 too.

```bash
$ python --version
Python 3.8.5
```
```bash
$ pip freeze
beautifulsoup4==4.9.3
google==3.0.0
grpcio==1.37.1
grpcio-tools==1.37.1
imageio==2.9.0
numpy==1.20.3
Pillow==8.2.0
protobuf==3.17.0
pypylon==1.7.2
Pyro5==5.12
serpent==1.40
six==1.16.0
soupsieve==2.2.1
```

thiesmoeller, 2021-06-14:
One hint to get more debugging info is to enable python memory debugging

Set `PYTHONMALLOC=debug` in environnement

meredithheskethfortescue, 2021-06-14:
I just tried it with Python 3.9.5 but I get the same errors. Are you on a Raspberry Pi? Because as mentioned earlier I can run it on a Windows PC too without any issues.

Your hint concerning `PYTHONMALLOC=debug`  actually got me a more verbose output. The output for the second example stays the same. But when I running it with `python -X dev` I get for the first one this:

```bash
free(): invalid pointer
Fatal Python error: Aborted

Current thread 0xb6f8b010 (most recent call first):
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1173 in create_module
  File "<frozen importlib._bootstrap>", line 565 in module_from_spec
  File "<frozen importlib._bootstrap>", line 666 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1058 in _handle_fromlist
  File "/root/camera_addon/venv/lib/python3.9/site-packages/grpc/__init__.py", line 23 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 855 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "<stdin>", line 1 in <module>
```

thiesmoeller, 2021-06-14:
There is quite some discussion on the grpc tracker about this issue

https://github.com/grpc/grpc/issues/26279

Please check how they solved this. Most probably you have to upgrade your grpc dependency

meredithheskethfortescue, 2021-06-15:
Thank you very much! This will probably be the solution.

closed 2021-06-15 by meredithheskethfortescue