basler/pypylon #374: grapping images by triggering regardless of time

ridvanozdemir, 2021-06-10:
how can i grap images by triggering regardless of time?

I get error when i wait
TimeoutException: Grab timed out. The acquisition is not started. : TimeoutException thrown (file 'instantcameraimpl.h', line 1056)

thiesmoeller, 2021-06-10:
Please check the hardware triggering notebook in https://github.com/basler/pypylon-samples

Either using the background loop or using wait forever.

ridvanozdemir, 2021-06-11:
thank you for fast answer

closed 2021-06-11 by ridvanozdemir

ridvanozdemir, 2021-07-28:
I checked the example but it is about software trigger,
Do you have any for hardware trigger?

thiesmoeller, 2021-07-28:
Take a look into our companion repository with more samples and notebooks that demonstrate most of pypylon features: https://github.com/basler/pypylon-samples