basler/pypylon #375: CreateDevice suddenly not working

rudi-mac, 2021-06-11:
I am operating two Basler USB cameras on a linux system using an InstantCameraArray (as described in here: https://github.com/basler/pypylon/blob/master/samples/grabmultiplecameras.py) since about 8 months without major problems.

Suddenly the device creation method in tlFactory is not executing without throwing an error. The code runs until the line:

andreasgau, 2021-06-14:
This is probably caused by an issue with the interprocess locks of the GenICam cache, like reported by issue https://github.com/basler/pypylon/issues/352. Your process cannot acquire the cache lock.

there are 2 Workarounds:
reboot the system.
link the genicam cache directory to a directory that is not existing. e.g GENICAM_CACHE_V3_1=/tmp/tmp2

rudi-mac, 2021-06-15:
Thanks a lot for the fast reply. Indeed, the problem has been resolved after complete system restart.

closed 2021-06-15 by rudi-mac

namranpan, 2022-09-19:
@andreasgau How do I use the second workaround? "link the genicam cache directory to a directory that is not existing. e.g GENICAM_CACHE_V3_1=/tmp/tmp2". I do not have a directory such as "GENICAM_CACHE_V3_1" like you have mentioned. 

What I can see is /home/user/genicam_xml_cache. Even after deleting this directory, running code re-creates this directory. The error still shows up in the CreateFirstDevice function.

andreasgau, 2022-09-19:
The environment variable GENICAM_CACHE_V3_1=/home/user/genicam_xml_cache is automatically set by pylon for a process using pylon if GENICAM_CACHE_V3_1 does not exist. If you set GENICAM_CACHE_V3_1 pointing to a non-existing directory before starting your application then this step is omitted and the cache will turned be off. Unfortunatly this is currently the only way to turn the GenICam cache off.

To set an environment variable once, use the export command in the prompt, not in a shell script and then start your application:
$ export GENICAM_CACHE_V3_1=/tmp/tmp2
$ .myapplication

namranpan, 2022-09-19:
Thanks for your help @andreasgau . However I am still facing the same issue.
The code is run from a python script.

Error trace:
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
  File "/home/adlink/.local/lib/python3.6/site-packages/pypylon/pylon.py", line 1592, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: Failed to open device '2676:ba03:2:1:2' for XML file download. Error: 'Failed to read the first register (maximum device response time). This can be caused by problems with the USB hardware used, i.e., the USB host controller and its corresponding driver  : RuntimeException thrown (file 'PylonUsbTL.cpp', line 487)

andreasgau, 2022-09-19:
This XML download step is located before the caching mechanism. Try powercycling the device (unplug and plug it in again). Could be a problem with insuffient power supply of the camera.

andreasgau, 2022-09-21:
Power supply, cable could be the problem. Please contact the Basler support directly regarding this issue. The Basler support has more information regarding hardware than me.

kishuinvain1, 2023-03-22:
> Thanks for your help @andreasgau . However I am still facing the same issue. The code is run from a python script. I am using an NVIDIA Jetson Xavier system.
> 
> Error trace: camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice()) File "/home/adlink/.local/lib/python3.6/site-packages/pypylon/pylon.py", line 1592, in CreateFirstDevice return _pylon.TlFactory_CreateFirstDevice(self, *args) _genicam.RuntimeException: Failed to open device '2676:ba03:2:1:2' for XML file download. Error: 'Failed to read the first register (maximum device response time). This can be caused by problems with the USB hardware used, i.e., the USB host controller and its corresponding driver : RuntimeException thrown (file 'PylonUsbTL.cpp', line 487)

@namranpan 
Is this problem solved, I'm facing more or less the same sort of problem. "Failed to download respective xml file"

namranpan, 2023-09-19:
@kishuinvain1 No, we ultimately had to change the cable