basler/pypylon #378: Distortion correction

sidW21, 2021-06-16:
Hi all,

This maybe a general CV question than a pypylon question, so apologies in advance. I'm currently using a Basler aca5472-17um unit with pypylon. Two aspects for which I wish to make corrections for grabbed images are: 1) vignetting correction 2) barrel distortion correction.

For vignetting correction, I was able to use the Pylon viewer to make the necessary adjustments as discussed [here](https://docs.baslerweb.com/vignetting-correction-(pylon-viewer)). I then saved the settings in a .pfs file, and every time I call the camera using pypylon, I load up the .pfs file, which enables the vignetting correction. 

I could not locate any information on whether there is an analogous solution available in pypylon or pylon viewer for barrel distortion corrections using a checkerboard pattern. There are a number of solutions from OpenCV as outlined [here](https://medium.com/vacatronics/3-ways-to-calibrate-your-camera-using-opencv-and-python-395528a51615), so my strategy currently is to use a routine outlined here to perform a post-processing distortion correction step.

Would anyone have any guidance on whether there is a more pypylonic way to handle a distortion correction? Would welcome any thoughts would be most welcome, and thank you all for the wonderful work in maintaining this resource!

SMA2016a, 2021-07-19:
the camera does not distortion corrections feature in the firmware. So you need to use OpenCV.

sidW21, 2021-07-21:
Thank you @SMA2016a for the clarification. As you mentioned, I ended up using an OpenCV routine as shown [here](https://automaticaddison.com/how-to-perform-camera-calibration-using-opencv/) and it was a straightforward process to complete. I hope this link proves useful for anyone else getting started. I'm closing this issue for now.

closed 2021-07-21 by sidW21