basler/pypylon #379: No devices can be found

keck-in-space, 2021-06-21:
Hello,

I am trying to use the pypylon library to enumerate the devices that are connected via Ethernet. However, when I call the following code, the `tlFactory.EnumerateDevices()` call is an empty tuple and the line to create the device fails with an exception. I've copied the code and the result below. Pylon sees 10 cameras connected.

```
from pypylon import pylon
from pypylon import genicam

tlFactory = pylon.TlFactory.GetInstance()
devices = tlFactory.EnumerateDevices()
print(f"devices: {devices}")
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
```

```
devices: ()
Traceback (most recent call last):
  File "C:\python_issue.py", line 7, in <module>
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
  File "C:\Users\USER\AppData\Local\Programs\Python\Python39\lib\site-packages\pypylon\pylon.py", line 1582, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'tlfactory.cpp', line 694)
```

Is there something else I need to do to let pypylon know that the devices are connected via Ethernet? Pylon itself sees them just fine.

Thank you!

Elektri4ker, 2021-07-02:
I have the same problem.
I've investigated it and can tell that `tlFactory.EnumerateDevices()` does not send broadcast discovery requests to '255.255.255.255'.
Here follows a picture of Wireshark dump while `tlFactory.EnumerateDevices()` was called. You see that it does broadcast only for a particular subnet (address 10.10.11.255)
[image]

@keck-in-space So your code will work if you manually configure devices with Pylon to appropriate ips and subnets.

Here follows a picture of Wireshark dump while Pylon IP configurator was working.
[image]

Is there a way to discover cameras with 255.255.255.255 broadcast, as Pylon does it?

thiesmoeller, 2021-07-02:
see here [gige ip config samples](https://github.com/basler/pypylon/blob/master/samples/gige_ip_config.py)

Elektri4ker, 2021-07-05:
> see here [gige ip config samples](https://github.com/basler/pypylon/blob/master/samples/gige_ip_config.py)
> 
> .. overall your camera should be in the same subnet as your host,
> to simplify the the whole setup.

Yes, I'm using this example. After calling`tlFactory.EnumerateDevices()` it ends up with "no GigE device found".

But can I somehow configure any connected device, even if it is not configured yet? Just like "Pylon IP configurator" does it via broadcast.

MattsProjects, 2021-10-08:
Yes, use Enumerate**All**Devices()
This is what the IPConfigurator uses.

milancurcic, 2022-05-07:
Same problem here. I have a GigE camera (acA2500-20gm) configured on the same subnet as the host. I can connect to it from Pylon Viewer, and I can ping it from the command line. However, ` tl_factory.EnumerateDevices()` keeps returning an empty tuple. 

@thiesmoeller, how do you recommend to best diagnose this? I'm using the pypylon from the latest commit on the master branch.

Thank you!

milancurcic, 2022-05-08:
FWIW what works for me is to not rely on `EnumerateDevices` but to connect to the camera directly via its IP address:

```python
tlf = pylon.TlFactory.GetInstance()
tl = tlf.CreateTl('BaslerGigE')
cam_info = tl.CreateDeviceInfo()
cam_info.SetIpAddress('192.168.2.100')
cam = pylon.InstantCamera(tlf.CreateDevice(cam_info))
```

thiesmoeller, 2022-05-09:
Make sure that the netmasks of the cameras really match the netmasks of your hosts Ethernet port.
Otherwise they won't react on the broadcast messages send out during enumeration.

thiesmoeller, 2022-07-06:
Hi @ghazalehtrb,

Please describe your setup:
Operation system and how your camera is connected.

What did you configure how the camera should get its IP address?

What is "the software" that works?

ghazalehtrb, 2022-07-06:
> Hi @ghazalehtrb,
> 
> Please describe your setup: Operation system and how your camera is connected.
> 
> What did you configure how the camera should get its IP address?
> 
> What is "the software" that works?

Thank you for your response @thiesmoeller ,
I have a Basler a2A1920-51gcPRO camera connected to an ethernet switch. My PC is also connected to the same switch. the operating system is Ubuntu 20.04. I am able to record with the pylon viewer software, but not with the pypylon api. when I try to configure the IP address with gige_ip_config.py I get the following error:

```
Traceback (most recent call last):
  File "gige_ip_config.py", line 38, in <module>
    raise EnvironmentError("no GigE device found")
OSError: no GigE device found
```
I have tried putting the camera on the same subnet with the pylon IP configurator but it doesn't change the error. 
I was able to make it work by playing around with the pylon IP configurator yesterday but I don't know how exactly. I keep having these issues every time I connect the camera to the switch and I don't know what the exact solution is.  

currently, this is the setup in the pylon IP configurator. I am able to record with the camera from the Pylon viewer software with this setup. However, the subnet mask of the camera (255.255.0.0) is different than the one shown for my PC ethernet connection (255.255.255.0). 
[image]

When I force it to be on the same subnet, it says the camera is unreachable, and I get the same error with the gige_ip_config.py .

[image]

could you please help me out with understanding the issue, I'm not very familiar with these concepts. 
Thanks

thiesmoeller, 2022-07-08:
Hi @ghazalehtrb, 

in the setup you describe it shouldn't be required to use this special procedure to open a camera by it's IP address.
Or has your PC multiple network interfaces which run in LLA mode?  ( LLA == network 169.254.x.x)

Have you followed the documentation for installing on gige?
https://docs.baslerweb.com/network-configuration-(gige-cameras)

closed 2022-07-12 by thiesmoeller

SMA2016a, 2022-07-22:
<your PC multiple network interfaces which run in LLA mode>  it is not allowed to run multiple camera is LLA Mode. Since all network devices will be in the same network,

swanandkAC, 2024-03-05:
> FWIW what works for me is to not rely on `EnumerateDevices` but to connect to the camera directly via its IP address:
>
> ```python
> tlf = pylon.TlFactory.GetInstance()
> tl = tlf.CreateTl('BaslerGigE') [...]

Thanks for the answer but after setting the IP and creating cam object, how do you read frames? I tried

```
def read(self):
        grabResult = cam.GrabOne(1000, pylon.TimeoutHandling_ThrowException)
        if grabResult.GrabSucceeded():
            img = grabResult.Array
            grabResult.Release()
            return img
```
But this gives an error saying

_genicam.TimeoutException: Grab timed out. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1361)

Can u pls help me out?