basler/pypylon #380: Reset camera features to factory settings

batrlatom, 2021-06-22:
Hello. Is it possible to reset features to some default state?

thiesmoeller, 2021-06-22:
https://docs.baslerweb.com/user-sets.html

And see also the in depth examples at
https://github.com/basler/pypylon-samples
There we also reset the cameras to default config...

thiesmoeller, 2021-06-29:
Not the samples folder in this repo but the pypylon-samples repo that I referenced in my last comment.

batrlatom, 2021-06-30:
I have found it at https://github.com/basler/pypylon-samples/blob/main/notebooks/grabstrategies.ipynb . Thanks!

closed 2021-06-30 by batrlatom

Mahdiekteshafi, 2024-05-10:
Hello, my camera is Basler acA2500 14um, but the resolution settings are manipulated and it shows 1920*1080 and I cannot set the resolution to its standard mode of 2592*1944.

thiesmoeller, 2024-05-11:
Please check the answer here https://github.com/basler/pypylon/issues/380#issuecomment-866079211

SMA2016a, 2024-05-11:
Set offsetx and offsetY to minimum then try to set the width and height