basler/pypylon #386: how to continuously grab and save images using basler ace cameras , without giving count of images, and the loop should be keep on running until that variable is true

thasnimolvs, 2021-07-10:


thiesmoeller, 2021-07-11:
start unlimited grabbing using: `cam.StartGrabbing()` you can stop the grabbing, when you detect your condition by calling `StopGrabbing()`

thasnimolvs, 2021-07-12:
@thiesmoeller 
currently am using threading method for this camera, so i was using while true loop before 1st class, and run the same code with while loop in main body also.is it work well or should i apply the while loop in thread created for grabbing, saving images and to the final main body also?

ToKraTheSecond, 2021-08-02:
@thasnimolvs check this example with grab strategies ;] https://github.com/basler/pypylon-samples/blob/main/notebooks/grabstrategies.ipynb

ToKraTheSecond, 2021-08-02:
@thasnimolvs  this two examples may also help you https://github.com/basler/pypylon/blob/master/samples/grabusinggrabloopthread.py https://github.com/basler/pypylon/blob/master/samples/grabstrategies.py

sdbhosale, 2023-09-20:
> start unlimited grabbing using: `cam.StartGrabbing()` you can stop the grabbing, when you detect your condition by calling `cam.StopGrabbing()`

How do I start 'limited grabbing'? I am trying to avoid calling cam.StopGrabbing. So is there a way to limit how long StartGrabbing stays active.

thiesmoeller, 2025-12-15:
This question about continuous grabbing and saving has been addressed in the comments with code examples. We are closing this issue. For further questions on continuous acquisition, please refer to the discussion or open a new issue.

closed 2025-12-15 by thiesmoeller