basler/pypylon #387: Pylon library initialization reference count is not zero, a call to PylonTerminate() is missing.

MarkusPTB, 2021-07-15:
I use pypylon Version: 1.7.2 on a windows computer in a GUI camera application programmed in
QT Creator version 4.13.3 based on QT 5.15.2.
I employed pip for the installation of pypylon.

The camera application runs fine, i can open the camera, set and read parameters and grab images.
However, whenever i quit the application the following warning appears:

"Pylon library initialization reference count is not zero, a call to PylonTerminate() is missing."

The warning already appears from just calling
"from pypylon import pylon"
without creating a pylon object.
Therefore, i would assume, it is caused by an issue in the precompiled library.

I would appreciate any suggestions how to get rid of the warning or how to resolve the issue.
Best regards,
Markus

SMA2016a, 2021-07-15:
Hi Marcus,

would you share your code, please?   or make a small script so I can easily see the issue.

MarkusPTB, 2021-07-16:
Sure, I created a minimal example project, which may be executed in Qt  creator.
With Python 3.8 installed.

Here is how to get it running from scratch:
1. Setup Qt Creator using the offline installer or creating a free Qt ID and using the online installer.
Offline installer is used to skip QT registration
(https://www.qt.io/offline-installers
Filename: qt-opensource-windows-x86-5.12.10.exe
Size: 3.7G (3964530304 bytes)

2. Install Python 3.8 (https://www.python.org/downloads/release/python-386/
Windows x86-64 executable installer)
In Qt creator make sure that Extras/Einstellungen/Python the right path is choosen for the python executable,
since the python install path may differ choosing install for a specific user or all users.

3. Use pip from the comand window to install Pyside2 and pypylon packages.
Pyside 2 is the interface, enabling the use of the Qt GUI elements in python and needs to be installed
(https://doc.qt.io/qtforpython/gettingstarted.html)
pip install pyside2

If you already have QT creator, python, Pyside2 and the pypylon package installed you may immediately run the project in the QT creator (I also have the pylon viewer software from basler installed on my system, but don't know if this is required). The example just contains a minimal GUI with 2 buttons and imports the pylon library. This is sufficient to recieve the warning, when closing the application. 
There is also a notification considering the default logger: This may get fixed by setting the environment variable
GENICAM_LOG_CONFIG_V3_1 to the path of the DebugLogging.properties file (attached (have to rename it to *.properties
[DebugLogging.txt](https://github.com/basler/pypylon/files/6829144/DebugLogging.txt)
) or
https://www.baslerweb.com/en/sales-support/knowledge-base/frequently-asked-questions/how-to-enable-pylon-logging/29829/ )

Attached example:
[minimal_code_example.zip](https://github.com/basler/pypylon/files/6829193/minimal_code_example.zip)

I have tested it and recieved the message
"Pylon library initialization reference count is not zero, a call to PylonTerminate() is missing."
 employing
QT Creator version 4.13.3 based on QT 5.15.2.
as well as the updated version
QT Creator version 4.15.2 based on QT 5.15.2.

Best regards
Markus

SMA2016a, 2021-07-17:
Hi Marcus,

I  had a quick look. You just call import nothing else. Is it correct?

 your code is running fine on a PC with Ubuntu 20.

in next step I will try to test your code  on windows and get back to you with the result

MarkusPTB, 2021-07-19:
Yes, this is correct. In my actual application I call functions from the pypylon library and my camera grabber runs.
However, to get the output:
"Pylon library initialization reference count is not zero, a call to PylonTerminate() is missing."
I only need to include pylon, without calling it.
[image]

SMA2016a, 2021-07-19:
obviously, the PylonterTerminate()  will not be called, please call pylon.PylonTerminate() before app.excc_()

MarkusPTB, 2021-07-19:
I have tried this already, but the call to "pylon.PylonTerminate()" causes the 
error:
AttributeError: module 'pypylon.pylon' has no attribute 'PylonTerminate'

As far as I can judge from the documentation (in the pylonviewer helpfiles), "PylonTerminate" is only defined in the underlying C++ libraries, the pypylon
package is build upon, but there appears to be no python equivalent for PylonTerminate().
Pylon terminate is referenced in the C++ documentation together with:
"
Pylon::PylonAutoInitTerm Class Reference
Helper class to automagically call PylonInitialize and PylonTerminate in constructor and destructor.
"

Based on this, I made my innitial assumption, that PylonTerminate() is missing somewhere in the C or C++ code, from which the pypylon package is build.

SMA2016a, 2021-07-19:
Hi Marcus,

the problem is, they are using Swig swig has #int() but not terminate. I think this is just a warning, not a serious issue.

MarkusPTB, 2021-07-19:
Yet, calling PylonTerminate() is generally the right idea.
I found a work around, that fixes the problem.

It is possible to use "ctypes", to load the "PylonC_v6_1.dll (64-Bit)"
into your python application. This DLL contains PylonTerminate() and you may call it to properly terminate the pylon application.

The code required is (assuming the DLL file is copied to the project folder):

from ctypes import cdll
Pylonlib = "./PylonC_v6_1.dll"
libc = cdll.LoadLibrary(Pylonlib)
libc.PylonTerminate()

I attached the running project including the 
[Minimal_Pylon_bug_example.zip](https://github.com/basler/pypylon/files/6840942/Minimal_Pylon_bug_example.zip)
PylonC_v6_1.dll (64-Bit) from the Basler pylon 6 SDK as well as the updated python script.
(Ctypes package has to be installed via pip for this to work.)

Thank you for the discussion and suggestions.
With the ctypes workaround I can solve the problem.
However, there most probably is a PylonTerminate() missing somewhere in the pypylon code, which causes the issue.

closed 2021-07-19 by MarkusPTB

SMA2016a, 2021-07-19:
Thanks for the update.