basler/pypylon #389: how to use pypylon in a docker on windows?

mashuai191, 2021-07-18:
the following code doesn't work in docker on Windows platform as docker cannot access the network of its host.

```
        tl_factory = pylon.TlFactory.GetInstance()  # 实例化相机
        devices = tl_factory.EnumerateDevices()
```

looks the same issue as https://github.com/basler/pypylon/issues/123, but with slight difference of i'm running docker on windows platform. On Linux there is already a solution (--network host) for this, but i didn't find a solution for windows as windows doesn't support the '--network host' parameter. Anyone know a solution or workaroud for this?

SMA2016a, 2021-07-18:
First try to forward the control channel ports to docker.   This will help to you EnumerateDevices(). . 
-p 3956:3956/udp
Pylon SDK need further 2  ports  > 45000 for streaming purpose.  This port can also be set for a fixed port number using pylon sdk.

camera.StreamGrabber.DestinationPort.SetValue(0); 

It may be easier for you cerate a ipvlan  instead of forwarding single ports.

mashuai191, 2021-07-19:
thanks @SMA2016a !, i will try that.

SMA2016a, 2021-07-23:
I have successfully tested  macvlan.... try this

SMA2016a, 2021-07-25:
IP of my camera :  192.168.3.11  connected to enp11s0f1  with IP : 192.168.3.10

so I add a vlan like this:

docker network  create  -d ipvlan    --subnet=192.168.3.0/24     --ip-range=192.168.3.0/24 -o ipvlan_mode=l2     -o parent=enp11s0f1 camvlan

docker run -ti --network=camvlan pylonimage

mashuai191, 2021-08-11:
the "Prerequisites" section says:

> Kernel requirements:
> To check your current kernel version, use uname -r
> IPvlan Linux kernel v4.2+ (support for earlier kernels exists but is buggy)

so looks this ipvlan driver is working only on Linux, and my docker host is Windows.

then i'm thinking about now that all the cameras in my env has configured fixed IP address, is there a way to replace the code "devices = tl_factory.EnumerateDevices()" with something like "devices = some_function(param: fixed_IP_list)"? if so, could you guide me the code snippet?
thanks!

ieren1503, 2022-07-22:
Hello everyone,

We use docker container for python project. We create all files and we need to connect basler camera via GigE vision protocol. We grab frames with pylon viewer very well. What can we do for access basler cam? 

we applied this code snipped;

"docker network create -d ipvlan --subnet=192.168.3.0/24 --ip-range=192.168.3.0/24 -o ipvlan_mode=l2 -o parent=enp11s0f1 camvlan

docker run -ti --network=camvlan pylonimage"

But did not access camera.  What can we do for access basler cam? 

Note: We use docker for desktop in windows and our docker container is including Ubuntu.

SMA2016a, 2022-07-27:
first of all, disable the ip filter in Ubuntu. refer the plyon's readme file.

if that does not help, create a TCP dump file on ubuntu system

ieren1503, 2022-07-27:
What is your host system? Have ever been try basler gige cam connection between windows host machine and docker ubuntu container? I tried your comments and nothing changed.