basler/pypylon #391: How to increase the quality of the images ?

abdullahfadel, 2021-07-21:
Hi, 

when I set the quality=100 in save_image.py and save the images  I see the best quality I get around 5MB only. When I try to capture the images in pylon Viewer I get 24MB quality of the image.  Do you know how to increase the quality of code or do we have any special parameters to add ?

Thanks

SMA2016a, 2021-07-22:
if you need an image with more quality then you should save the image uncompressed in png or bmp

abdullahfadel, 2021-07-22:
@SMA2016a Do you know how to save uncompressed images? I am following this example : 
https://github.com/basler/pypylon/blob/master/samples/save_image.py

SMA2016a, 2021-07-23:
filename = "saved_pypylon_img_%d.png" % i
      img.Save(pylon.ImageFileFormat_Png, filename)

simply remove the jpeg path.

thiesmoeller, 2025-12-15:
This question about image quality has been addressed. The quality setting in save_image.py refers to JPEG compression quality, not pixel format. For full quality, use PNG or TIFF format. We are closing this issue. For further questions on image formats, please refer to the Pylon SDK documentation.