basler/pypylon #392: when we connect 3 basler cameras working at the same time, is it possible to Let us say you have attached three cameras C1, C2 and C3 and connected to three ports P1, P2 and P3. If your code now says that their serial IDs are S1, S2, S3.  No matter to which port I connect C1 it should always be S1, similarly for C2 and C3.how to get it

thasnimolvs, 2021-07-23:


SMA2016a, 2021-07-23:
<No matter to which port I connect C1 it should always be S1>  No.

I would suggest you set user-defined name for each camera.  Let say  cam1,cam2,cam3  then open the camera using a predefined camera info object.
so you have always sorted list.

thasnimolvs, 2021-07-24:
@SMA2016a 
Currently am initialising and that looping is using enumerate function..I need every time a proper order of folders ..sir , could you please explain clearly. ?
Now am creating that folder based on their serial no

SMA2016a, 2021-07-25:
Please refer the sample
https://github.com/basler/pypylon/blob/master/samples/grabmultiplecameras.py

you just need to adapt the attach line.

# Create and attach all Pylon Devices.
    SNs= ["sn1", "sn2", "sn3"]
    for i, cam in enumerate(cameras):
         device_info = pylon.DeviceInfo()
         device_info.SetSerialNumber(SNs[i])
        cam.Attach(tlFactory.CreateDevice(device_info))

So you have always in same order.

thasnimolvs, 2021-07-25:
@SMA2016a 
Hi sir,
i have one more doubt, when i captured the images from multiple camera and analysed by using histogram of delta t (time difference between each time stamp) but the difference is much lesser means less than 1ms for 60 fps, so it seems to be some problem with code.
i think its due to multiple images are added to the buffer, but they are not  cleared at regular time 
and putting the timestamp only reading from the buffer , timestamp depends on whrn we read data , here by attaching my code 
here the images are transfred to an array. is it possible to modify this code that array part?????????????
###############################################################################
# Imports
###############################################################################
from pypylon import pylon
from pypylon import genicam
from datetime import datetime
import time
# Import OpenCV
import cv2
from PIL import Image
# Import os for creating directories.
import os
#os.environ["PYLON_CAMEMU"] = "2"
import sys
# Import shutil for deleting folders.
import shutil
# Import threading for creating threads.
import threading
import numpy as np
###############################################################################
# Parameters
###############################################################################
#countOfImagesToGrab=60
#fps=60
#outputDir = '/mnt/ramdisk/output'
#outputDir = '/mnt/Disk3/test/output'
#serial_numbers = ['40092332','40092327','40083731']
outputDir = '/mnt/Disk3/test/trial'
#outputDir = '/mnt/Dicccsk3/test/trial'
#outputDir = 'output'
###############################################################################
# Global variables
###############################################################################
  # Get the transport layer factory.
tlFactory = pylon.TlFactory.GetInstance()
# Get all attached devices and exit application if no device is found.
devices = tlFactory.EnumerateDevices()
print(devices)

class CameraImage:
    
    def __init__(self, device):
        self.runningStatus = False
        self.interruptStatus = False
        self.thread = None
        self.imageList = []
        self.device = device
        
    def frameGrabber(self):
        cam = pylon.InstantCamera(tlFactory.CreateDevice(self.device))
        cam.Open() 
        cam.Width= 1440
        cam.Height=1080
        cam.ExposureTime = 3000
        cam.TriggerSelector.SetValue("FrameBurstStart")
        cam.TriggerMode.SetValue("On") #hardware trigger
        cam.TriggerSource.SetValue("Line4")
        cam.TriggerActivation.SetValue('RisingEdge')
        #cam.TriggerSelector.SetValue("FrameStart") # FrameBurstStart is used for capturing a series of images with a single signal, please choose FrameStart if wish to have one signal one image
        #cam.TriggerDelay.SetValue(300)
        cam.PgiMode.SetValue("On")
        cam.NoiseReduction.SetValue(2.0)
        cam.SharpnessEnhancement.SetValue(3.984365)
       # cam.MaxNumBuffer=1000
[... 94 lines cut ...]

    #startTime = time.time()
        # Create the video object.
    cameraImage1 = CameraImage(devices[0])
    saveImage1 = SaveImage(cameraImage1)

    cameraImage2 = CameraImage(devices[1])
    saveImage2 = SaveImage(cameraImage2)

    cameraImage3 = CameraImage(devices[2])
    saveImage3 = SaveImage(cameraImage3)

    cameraImage1.startGrabbing()
    saveImage1.startSaving()
    cameraImage2.startGrabbing()
    saveImage2.startSaving()
    cameraImage3.startGrabbing()
    saveImage3.startSaving()

    cameraImage1.thread.join()
    saveImage1.thread.join()
    cameraImage2.thread.join()
    saveImage2.thread.join()
    cameraImage3.thread.join()
    saveImage3.thread.join()

SMA2016a, 2021-07-26:
would you also add your histro file, please

thasnimolvs, 2021-07-26:
@SMA2016a 
auto exposure function is already off .
how to set it in python??
[fullanalysis2.xlsx](https://github.com/basler/pypylon/files/6878409/fullanalysis2.xlsx)
[image]
this code is for histo

import os,csv
import openpyxl
import matplotlib.pyplot as plt
import numpy as np
f=open("value.xlsx","w")
#file1 = sort("weights.txt","w")#append mode 
#/home/thasni/CODE/samples/ThreadingExample
#w=csv.writer(filename)
def main():
    time=[]
    arrayTotTime=[]
    file=[]
    diffArray=[]
    xaxis=[]
    #for path, dirs, files in os.walk('/mnt/Disk3/test/trial'):
    fileName=os.listdir('/mnt/Disk3/test/trial') #give folder here
    print(fileName)

    for f in range (0,len(fileName)):
        time.append(fileName[f].split(".")[1]) #split & inserting timestamps to array
    print(time)

    for t in range (0,len(time)):
        hour=int(time[t].split("_")[0])*60*60*1000000 #hour to usec
        min=int(time[t].split("_")[1])*60*1000000 #min to usec
        sec=int(time[t].split("_")[2])*1000000 #sec to usec
        totalTimeInuSec = hour+min+sec+int(time[t].split("_")[3])
        print(totalTimeInuSec)
        arrayTotTime.append(totalTimeInuSec)#array of totalTimeInuSec
    
    arrayTotTime.sort()
    print(arrayTotTime) #all timestamp to millisecond and is inserting into an array

    wb = openpyxl.Workbook() #opening a new workbook
    sheet = wb.active

#finding delta
    sheet.cell(row=1, column=1).value = "Time Delta in uSec"
    print("Delta")
    for d in range (0,len(arrayTotTime)-1):
        diff = arrayTotTime[d+1]-arrayTotTime[d]
        diffArray.append(diff)
        xaxis.append(d+1)
        print(diff, "uSec")
        sheet.cell(row=d+2, column=1).value = diff #writing to the excel
   wb.save("/mnt/Disk3/test/value.xlsx") #saving the file as value.xlsx
    
    
    # plt.bar(xaxis, diffArray)
    # plt.ylabel("delta t in uSec")
    # plt.title("Title")#title for the plot

    plt.hist(diffArray,1,range=[0, 100])
    plt.show()
    
    ###############################################################################

if __name__ == "__main__":
  main()

thasnimolvs, 2021-07-26:
@SMA2016a 
is there any issues in settings??

thiesmoeller, 2025-12-15:
This issue about camera ordering has been addressed in the comments with guidance on using serial numbers to identify cameras. We are closing this issue. For further questions on camera identification, please refer to the discussion or open a new issue.

closed 2025-12-15 by thiesmoeller