basler/pypylon #393: Waiting for frame trigger ready times out

rdrandom, 2021-07-28:
Hello everyone, first of all, sorry for my English.
I have a problem when I run this script that I have trouble understanding because I start:

`import sys
import time

from pypylon import pylon

from imageeventprinter import ImageEventPrinter
from configurationeventprinter import ConfigurationEventPrinter

exitCode = 0
print('exitcode')

ip_address = '192.168.0.123:3956'
factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = ptl.CreateDeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = factory.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)
print('recupid')

camera.Open()
print(camera.Open())
print("DeviceClass: ", camera.GetDeviceInfo().GetDeviceClass())
print("DeviceFactory: ", camera.GetDeviceInfo().GetDeviceFactory())
print("ModelName: ", camera.GetDeviceInfo().GetModelName())
print("Grab using the GrabStrategy_OneByOne default strategy:")

camera.StartGrabbing(pylon.GrabStrategy_OneByOne)
print('the grabstrategy')
         
for i in range(3):
    print('debutboucle')
    if camera.WaitForFrameTriggerReady(5000, pylon.TimeoutHandling_ThrowException):
        print('avant : ',i)
        camera.ExecuteSoftwareTrigger()
        print('apres : ',i)
print('apresboucle')`

I am getting this prints and error:

exitcode
recupid
None
DeviceClass:  BaslerGigE
DeviceFactory:  GigE/BaslerGigE 6.1.1.18345
ModelName:  acA2040-25gc
Grab using the GrabStrategy_OneByOne default strategy:
the grabstrategy
debutboucle
Traceback (most recent call last):
  File "./test.py", line 74, in <module>
    if camera.WaitForFrameTriggerReady(5000, pylon.TimeoutHandling_ThrowException):
  File "/usr/local/lib/python3.8/dist-packages/pypylon/pylon.py", line 3927, in WaitForFrameTriggerReady
    return _pylon.InstantCamera_WaitForFrameTriggerReady(self, *args)
_genicam.TimeoutException: Waiting for frame trigger ready timed out. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1692)

Thank you in advance for any help you can give me!

rdrandom, 2021-07-28:
I add a trigger  : 
for i in range(3):
    print('debutboucle')
    camera.TriggerSource.SetValue("Line1")
    camera.TriggerMode.SetValue("On")
    print('apres trigger')
    if camera.WaitForFrameTriggerReady(5000, pylon.TimeoutHandling_ThrowException):
        print('avant : ',i)
        camera.ExecuteSoftwareTrigger()
        print('apres : ',i)
print('apresboucle')

it solve this problem but another appears...

_genicam.TimeoutException: Grab timed out. The acquisition is not started. : Tim                                                                      eoutException thrown (file 'InstantCameraImpl.h', line 1056)

SMA2016a, 2021-07-29:
you may try ip_address = '192.168.0.123'  without port number

rdrandom, 2021-07-29:
thanks but I tried and nothing has changed, still the same error ...

SMA2016a, 2021-07-29:
is camera working fine in pylonviewer or simple grab sample code?

rdrandom, 2021-07-30:
hey ! I use pylonviewer and i can have an image yes

SMA2016a, 2021-07-30:
try this.

empty_camera_info = pylon.DeviceInfo()

rdrandom, 2021-07-30:
if i use this : 
`python
ip_address = '192.168.0.123'
factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = pylon.DeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = factory.CreateDevice(empty_camera_info)
cam = pylon.InstantCamera(camera_device)

cam.Open()

cam.StartGrabbing(pylon.GrabStrategy_OneByOne)
`

i have this error : 

Traceback (most recent call last):
  File "./sauvegarde_image.py", line 24, in <module>
    camera_device = factory.CreateDevice(empty_camera_info)
  File "/usr/local/lib/python3.8/dist-packages/pypylon/pylon.py", line 1594, in CreateDevice
    return _pylon.TlFactory_CreateDevice(self, *args)
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)

rdrandom, 2021-08-03:
Good morning ! I discovered something!
When I run my script on PyCharm on the computer, 0 problem and everything is done correctly if I am connected by wire and not by wifi! If it's wifi: same problem as on the container!

So I tested ... I can connect to the camera with Pylon Viewer but not take pictures or have any image stream if I am in wifi! Only the wired that works.

Is this normal? Are there ways to change this?

SMA2016a, 2021-08-03:
now you have 2 cases in one thread. This makes supporting you a bit more complicated.If you discover new issue create a new question.

1)  it seems that you are not setting up the proper IP address in the info object.

ip_address = '*****'
factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = pylon.DeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = factory.CreateDevice(empty_camera_info)
cam = pylon.InstantCamera(camera_device)

cam.Open()

cam.StartGrabbing(pylon.GrabStrategy_OneByOne)

2) Most properly you waln componant does not perform well.

     Set the packet size in transport layer to 1500 and increase the inter packet delay in small steps.

rdrandom, 2021-08-03:
excuse me if it seems to you to be two different problems because I just wanted to explain my progress and I thought that might enlighten you on the problem. (in itself I'm not trying to make it work on pylon viewer!)

in fact, i don't understand why there would be a problem with the ip address since, when i'm wired, it doesn't pose a problem and i can recover photos.

when I am connected in wifi, I tried to increase the size of the inter packets but I had to go "too" fast because that did not change anything. I'm trying to do it more slowly, I'll keep you posted

thank you for helping me and sorry for any questions!