basler/pypylon #394: Cannot install pypylon on RPi.

Ilias9807, 2021-07-30:
I am trying to install the `pypylon` python wrapper for the pylon camera suite in my Raspberry pi 3 using `pip3` but I receive thre following error: 

[image]

My RPi runs on ARM Linux 32bit, I am using python 3.7.3 and I have the pylon camera suite 6.2.0 installed.
I also tried buliding it from source following the directions you provided but with no luck.

Any help would be much appreciated.

jackcary, 2021-07-30:
`pip3 install --upgrade pip`

Ilias9807, 2021-07-30:
@jackcary Thank you for the response. My pip is up to date with the latest version, and I already run an `install upgrade` just in case, but got nowhere. Regarding the rest of your suggestions I 'll try them out and get back to you.

jackcary, 2021-07-30:
@Ilias9807 I also had better luck when I specified the exact wheel file for pip to install.  In your case (32-bit ARM) that may be pypylon-1.7.2-cp37-cp37m-manylinux_2_24_armv7l.whl

`pip3 install https://github.com/basler/pypylon/releases/download/1.7.2/pypylon-1.7.2-cp37-cp37m-manylinux_2_24_armv7l.whl`

SMA2016a, 2021-07-31:
<My pip is up to date with the latest version>
waht pip version are you  using?

Ilias9807, 2021-08-03:
> waht pip version are you using?

@SMA2016a Uppon checking the pip version on my system is 20.3.4, so not the very last published version.

Ilias9807, 2021-08-03:
UPDATE: Apparently my pip3 version was quite outdated, (version 18.x.x), so running an update like so:

`sudo -H pip3 install --upgrade pip`

Did the trick.

closed 2021-08-03 by Ilias9807