basler/pypylon #395: how to save an basler camera image filename with its chunk time stamp  in python?

thasnimolvs, 2021-07-30:
i tried to save the basler usb camera image filename with timestamp, that time it is not accurate, so next i tried with chunk image timestamp. so i want to save each image filename withits corresponding timestamp

SMA2016a, 2021-07-30:
the time stamp in grabresult and chunk is the same. 

< that time it is not accurate>  would you explain in detail your statement,please?

thasnimolvs, 2021-07-31:
actually we need to analyse time between each timestamp,
so first we used datetime. now and time.time , but  both these cases when we analyse the delta t it is not a correct value,   so we go for chunk data timestamp ,
so while using the sample code in github _ grabchunkimage.py , we are getting a timestamp. 
1. i want to know its format...like hr, min, sec, milsec.. or else 
2.  datetime.now().strftime("%Y-%m-%d.%H_%M_%S_%f.-{}.tiff".format(serial_number)), earlier am using this using this format for filename so instead of that i want to use chunk timestamp of each corresponding images as its filename? 
3. how to set the filename in  python

SMA2016a, 2021-07-31:
in USB camera, the time stamp in chunk as well as in grabresult given in ticks = 1ns.

You need to convert ticks into Daytime.