basler/pypylon #396: First image from the camera takes time to store in buffer

vishalbende, 2021-08-03:
I am using basler linescan camera . With hardware trigger directly to camera and grabbing result is handled  with `pylon.ImageEventHandler`  handler. But if trigger then for the first image after camera is steady for 3-4 seconds it takes about 2-4 seconds to store image in buffer and to call image event handler

thiesmoeller, 2021-08-05:
Hi,

Assuming that you configured Linestart on the HW trigger and FrameStart with no trigger ?:

What is your trigger frequency/exposure time and virtual frame Height?

The image event handler gets signaled if the _complete_ virtual frame has been received.

vishalbende, 2021-08-12:
Hi, 

Frame start and line start are both configured on same trigger . 

Height - 1700 
Exposure - 2500

thiesmoeller, 2021-08-12:
Your exposure time will result in a frame time of 1700*2500us = 4.25 s frame time.

You only receive the virtual frames in pylon.
If you want to have shorten the frame time you would have to reduce the height, but then adjust your machine vision to handle this.