basler/pypylon #398: how to control the MaxCurrent of basler's light with Python (Pypylon)

mahmoudabderahman, 2021-08-13:
Hi all,

I am trying to change the value of Max Current for the light control using Python. Here is my code:

self.camera.BslLightDeviceMaxCurrent.SetValue(ctypes.c_float(200.0))

But I am getting the following error:

return _genicam.IFloat_SetValue(self, Value, Verify)
TypeError: Wrong number of type of arguments for overloaded function 'IFloat_SetValue'.
     Possible C/C++ prototypes are:
           GENAPI_NAMESPACE::IFloat::SetValue(double, bool)
           GENAPI_NAMESPACE::IFloat::SetValue(double)

I tried changing the parameter also to ctypes.c_double(200.0) and it also didn't resolve the problem.

Thanks in advance!

thiesmoeller, 2021-08-13:
Why not only:

```python
self.camera.BslLightDeviceMaxCurrent.SetValue(200.0)
```

mahmoudabderahman, 2021-08-13:
> Why not only:
> 
> ```python
> self.camera.BslLightDeviceMaxCurrent.SetValue(200.0)
> ```

Thanks for the reply!

I first tried this code and got the following error:

_genicam.AccessException: Node is not writable. : AccessException thrown in node 'BslLightDeviceMaxCurrent' while calling 'BslLightDeviceMaxCurrent.SetValue()' (file 'FloatT.h', line 85)

thiesmoeller, 2021-08-14:
Is your camera "open"?

`self.camera.Open()`

_before_ your access to the feature

mahmoudabderahman, 2021-08-16:
> Is your camera "open"?
> 
> `self.camera.Open()`
> 
> _before_ your access to the feature

Yes.

Here is the flow:

```
self.camera.Open()
self.set_camera_parameter()
```
In `self.set_camera_parameter()`, I do the configuration related to the light controller.

mahmoudabderahman, 2021-08-18:
The `BslLightDeviceOperationMode `should actually be "Off" before adjusting the Max Current. Turning the `BslLightDeviceOperationMode `off solved the problem. Sorry for the inconvenience.

closed 2021-08-18 by mahmoudabderahman