basler/pypylon #399: Compression Beyond Feature

OOmerKerem, 2021-08-17:
Hi all,

I have Basler Ace 2 Pro (a2A1920-51gcPRO). Firstly I used my camera with Pylon Viewer but I couldn't use a few feature like Compression Beyond, Scaling etc. When I contact with local distributor, they told me that recording features of Pylon Viewer are limited and ıf I want to these features, I have to write my own code. I started coding with pypylon package on Python but I was unable to use these features with the sample codes given on the https://docs.baslerweb.com/compression-beyond. Is there any other way of using these features with the pypylon package?

Thank you for your help in advance.

SMA2016a, 2021-08-18:
please refer the c++ cample code Utility_ImageDecompressor.cpp can be found in /opt/pylon/share/Samples/C++/Utility_ImageDecompressor

thiesmoeller, 2021-08-18:
this test is for USB, but basic principle of decompression setup is identical:
[pylon_tests/usb/decompressiontest.py](https://github.com/basler/pypylon/blob/master/tests/pylon_tests/usb/decompressiontest.py)

OOmerKerem, 2021-08-19:
Thanks for reply, but I can't see compressed image on image window.  I am attaching my code below.
```
from pypylon import pylon
from pypylon import genicam
import sys

exitCode = 0
MaxGrabbingImage = 1000

try:
    imageWindow = pylon.PylonImageWindow()
    imageWindow.Create(1)
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    camera.Open()
    print("Camera:", camera.GetDeviceInfo().GetModelName())
    
    camera.Width.SetValue(960)
    camera.Height.SetValue(600)
    print("Width:", camera.Width.GetValue())
    print("Height:", camera.Height.GetValue())
    
    camera.ImageCompressionMode = "BaslerCompressionBeyond"
    camera.ImageCompressionRateOption = "Lossless"
    camera.BslImageCompressionRatio = 30.0
 
    print("FPS:", camera.ResultingFrameRate.GetValue())
   
    camera.StartGrabbingMax(MaxGrabbingImage)

    while camera.IsGrabbing():
        grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

        if grabResult.GrabSucceeded():
            imageWindow.SetImage(grabResult)
            imageWindow.Show()
        else:
            print("Error: ", grabResult.ErrorCode, grabResult.Release)
        grabResult.Release()
    camera.Close()
    imageWindow.Close()
    

except genicam.GenericException as e:
    print("An exception occurred.")
    print(e.GetDescription())
    exitCode = 1;

sys.exit(exitCode)
```
and  I didn't find any sample about Scaling or Pixel Beyond features. If there any sample can you share it?

thiesmoeller, 2021-08-24:
Check line 60ff
You have to decompress every frame you get

OOmerKerem, 2021-08-25:
I add decompression part to my code like this:
```
    decompressor = pylon.ImageDecompressor()
    descriptor = camera.BslImageCompressionBCBDescriptor.GetAll()
    decompressor.SetCompressionDescriptor(descriptor)
   
    camera.StartGrabbingMax(MaxGrabbingImage)

    while camera.IsGrabbing():
        compressed_image = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        decompressed_image = decompressor.DecompressImage(compressed_image)

        if compressed_image.GrabSucceeded():
            imageWindow.SetImage(decompressed_image)
            imageWindow.Show()
        else:
            print("Error:", compressed_image.ErrorCode, decompressed_image.Release)
        decompressed_image.Release()
    camera.Close()
    imageWindow.Close()
```
but I take this error:
```
Python 3.8.10 (tags/v3.8.10:3d8993a, May  3 2021, 11:48:03) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> 
======= RESTART: C:\Users\...\Basler.py =======
Camera: a2A1920-51gcPRO
Width: 960
Height: 600
Compression: 61.4
FPS: 7.782282854852642
Traceback (most recent call last):
  File "C:\Users\...\Basler.py", line 45, in <module>
    imageWindow.SetImage(decompressed_image)
  File "C:\Users\...\AppData\Local\Programs\Python\Python38\lib\site-packages\pypylon\pylon.py", line 6968, in SetImage
    return _pylon.PylonImageWindow_SetImage(self, image)
TypeError: in method 'PylonImageWindow_SetImage', argument 2 of type 'Pylon::CGrabResultPtr const &'
>>> 
```
I also try grabbing with "with camera.GrabOne(10000) as compressed_image" but nothing changed.

closed 2024-05-09 by SMA2016a