basler/pypylon #401: Image Grabbing with multiple GigE Cameras and Hardware Trigger

jsiebert, 2021-08-18:
Hey there,

I'm trying to grab images from multiple GigE Cameras where the grabbing itself should be triggered by a hardware trigger (only one frame per trigger). I'm running my script on Ubuntu 20.04 LTS with Python 3.8.10 and PyPylon 1.7.2.

The code is as follows:
```python
try:
    tl_factory = pylon.TlFactory.GetInstance()
    devices = [dev for dev in tl_factory.EnumerateDevices() if dev.GetDeviceClass() == "BaslerGigE"]

    if len(devices) == 0:
        print("No cameras present.")
    else:
        for dev in devices:
            print(f"{dev.GetModelName()}@{dev.GetIpAddress()}")

        cameras = pylon.InstantCameraArray(min(len(devices), 10))
        for i, cam in enumerate(cameras):
            cam.RegisterConfiguration(ConfigurationEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
            cam.RegisterImageEventHandler(ImageEventHandler(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
            cam.Attach(tl_factory.CreateDevice(devices[i]))
            cam.Open()
            cam.AcquisitionMode.SetValue('SingleFrame')
            cam.TriggerMode.SetValue("On")
            cam.TriggerSource.SetValue('Line1')
            cam.TriggerSelector.SetValue("FrameStart")
            cam.TriggerActivation.SetValue('RisingEdge')
            cam.TriggerDelayAbs.SetValue(0.0)
            cam.LineSelector.SetValue('Line1')
            cam.LineMode.SetValue('Input')
            cam.LineDebouncerTimeAbs.SetValue(10.0)
            cam.LineFormat.SetValue('OptoCoupled')

        cameras.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)

except genicam.GenericException:
    print("An exception occurred during camera initialization.")
    sys.exit(1)
```
The problem that occurs now is that right after the cameras start grabbing, the grabbing is stopped again and all camera handles are closed and destroyed. Here is the output from the script:
```shell
acA3088-16gm@192.168.51.11
acA3088-16gm@192.168.51.12
acA3088-16gm@192.168.51.13
acA3088-16gc@192.168.52.11
acA3088-16gc@192.168.51.10
acA3088-16gc@192.168.52.12
acA3088-16gc@192.168.52.13
acA3088-16gc@192.168.52.10
OnAttach event
OnAttached event for device  acA3088-16gm
OnOpen event for device  acA3088-16gm
OnOpened event for device  acA3088-16gm
OnAttach event
OnAttached event for device  acA3088-16gm
OnOpen event for device  acA3088-16gm
OnOpened event for device  acA3088-16gm
OnAttach event
OnAttached event for device  acA3088-16gm
OnOpen event for device  acA3088-16gm
OnOpened event for device  acA3088-16gm
OnAttach event
OnAttached event for device  acA3088-16gc
OnOpen event for device  acA3088-16gc
OnOpened event for device  acA3088-16gc
OnAttach event
OnAttached event for device  acA3088-16gc
OnOpen event for device  acA3088-16gc
OnOpened event for device  acA3088-16gc
OnAttach event
OnAttached event for device  acA3088-16gc
OnOpen event for device  acA3088-16gc
OnOpened event for device  acA3088-16gc
OnAttach event
OnAttached event for device  acA3088-16gc
OnOpen event for device  acA3088-16gc
OnOpened event for device  acA3088-16gc
OnAttach event
OnAttached event for device  acA3088-16gc
OnOpen event for device  acA3088-16gc
OnOpened event for device  acA3088-16gc
OnGrabStart event for device  acA3088-16gm
OnGrabStarted event for device  acA3088-16gm
OnGrabStart event for device  acA3088-16gm
OnGrabStarted event for device  acA3088-16gm
OnGrabStart event for device  acA3088-16gm
OnGrabStarted event for device  acA3088-16gm
OnGrabStart event for device  acA3088-16gc
OnGrabStarted event for device  acA3088-16gc
OnGrabStart event for device  acA3088-16gc
OnGrabStarted event for device  acA3088-16gc
OnGrabStart event for device  acA3088-16gc
OnGrabStarted event for device  acA3088-16gc
OnGrabStart event for device  acA3088-16gc
OnGrabStarted event for device  acA3088-16gc
OnGrabStart event for device  acA3088-16gc
OnGrabStarted event for device  acA3088-16gc
OnGrabStop event for device  acA3088-16gm
OnGrabStopped event for device  acA3088-16gm
OnGrabStop event for device  acA3088-16gm
OnGrabStopped event for device  acA3088-16gm
OnGrabStop event for device  acA3088-16gm
OnGrabStopped event for device  acA3088-16gm
OnGrabStop event for device  acA3088-16gc
OnGrabStopped event for device  acA3088-16gc
OnGrabStop event for device  acA3088-16gc
OnGrabStopped event for device  acA3088-16gc
OnGrabStop event for device  acA3088-16gc
OnGrabStopped event for device  acA3088-16gc
OnGrabStop event for device  acA3088-16gc
OnGrabStopped event for device  acA3088-16gc
OnGrabStop event for device  acA3088-16gc
OnGrabStopped event for device  acA3088-16gc
OnClose event for device  acA3088-16gc
OnClosed event for device  acA3088-16gc
OnDestroy event for device  acA3088-16gc
OnDestroyed event
OnClose event for device  acA3088-16gc
OnClosed event for device  acA3088-16gc
OnDestroy event for device  acA3088-16gc
OnDestroyed event
OnClose event for device  acA3088-16gc
OnClosed event for device  acA3088-16gc
OnDestroy event for device  acA3088-16gc
OnDestroyed event
OnClose event for device  acA3088-16gc
OnClosed event for device  acA3088-16gc
OnDestroy event for device  acA3088-16gc
OnDestroyed event
OnClose event for device  acA3088-16gc
OnClosed event for device  acA3088-16gc
OnDestroy event for device  acA3088-16gc
OnDestroyed event
OnClose event for device  acA3088-16gm
OnClosed event for device  acA3088-16gm
OnDestroy event for device  acA3088-16gm
OnDestroyed event
OnClose event for device  acA3088-16gm
OnClosed event for device  acA3088-16gm
OnDestroy event for device  acA3088-16gm
OnDestroyed event
OnClose event for device  acA3088-16gm
OnClosed event for device  acA3088-16gm
OnDestroy event for device  acA3088-16gm
OnDestroyed event
```
Does anybody have a clue why this does happen? I've seen many other examples here where similar code seemed to work for other people...

Thanks in advance

Jens

thiesmoeller, 2021-08-18:
`cam.AcquisitionMode.SetValue('SingleFrame')`

This has to be "Continuous".
With the FrameStart trigger you already make sure to get only one frame per trigger

jsiebert, 2021-08-18:
Hey @thiesmoeller,

thanks for your fast reply! Unfortunately, changing the AcquisitionMode to "Continuous" doesn't solve the problem.

thiesmoeller, 2021-08-19:
Took a moment to realize ;-)

You exit your program.
What you see is the shutdown of pylon.

StartGrabbing is only a 'trigger' to start the grab engine.
You have to keep your program running after it.
At minimum e.g. a sleep ...

jsiebert, 2021-08-19:
Ha! Thanks @thiesmoeller for the hint! 

While it wasn't the root cause for the error (I already did a call to ```sleep``` in a loop in the main function), the call to ```StartGrabbing``` was part of the initialization function where other things happen between the call to ```StartGrabbing``` and actually starting the main loop.
Now, I've moved the call to ```StartGrabbing``` closer to the main loop. Et violà: now it doesn't stop the grabbing!

Thanks again! Closing this issue...

closed 2021-08-19 by jsiebert